motioneye-project/motioneyeos #931: raspberry pi zero : wpa_supplicant configuration broken after parameter modified in admin interface

VincentGaudeul, 2017-05-06:
Hi again. 
I've got another problem here. 
Using motioneyeos-raspberrypi-20170329 image on a raspberry pi zero W. 
Everything is ok until I rename surveillance user username (set to hirooekimae) and password in admin interface (I do not want to use admin user for everyday surveillance).  
System reboots and then wpa_supplicant fails... and system reboots once more. 

Any guess ?

VincentGaudeul, 2017-05-06:
Actually, it seems that key_mgmt is set to NONE in wpa_supplicant.conf file when some parameters are modified in admin interface (even if not related to network configuration). This seems to break system.

VincentGaudeul, 2017-05-06:
I found a workaround for tonight, by creating symbolic links to /boot/wpa_supplicant.conf : 

ln -s /boot/wpa_supplicant.conf /data/etc/wpa_supplicant.conf
ln -s /boot/wpa_supplicant.conf /etc/wpa_supplicant.conf

It seems to work.

JohBod, 2017-05-07:
Hi,

I'm running MotionEyeOS on two Rpi Zero W and became a little worried by your message about loss of WiFi configuration. In the admin web interface, I have switched FTP server, Samba server and SSH server on and off in various combinations but I never lost the WiFi config.

This is how I prepare my SD cards:
- Write motioneyeos-raspberrypi-20170329.img.gz to the SD card with Etcher on a windows PC
- Remove the ("safely ejected") card and re-insert it.
- Write a suitable wpa_supplicant.conf file to the root of the boot partition.
- Put SD card in Rpi, boot, login and do remaining setup via the web interface.

This appears to work flawlessly but I was still a bit worried by your problem. I tried your suggestion to modify /etc/init.d/S35Wifi and made all three references to wpa_supplicant.conf point to /boot/wpa_supplicant.conf. After rebooting, my Rpi0zW did connect to the router via DHCP but got IP 0.0.0.0 (!) according to the router's WLAN association table. It was of course impossible to connect to this IP from a web browser.

Do I really have to worry about losing WiFi config by messing with the web interface or should I somehow try to make your /etc/init.d/S35Wifi modification work?

Maybe Calin can chime in and give a hint?

Regards
Johan

VincentGaudeul, 2017-05-07:
Hi @JohBod 
Strange issue. I do prepare my SD cards as you do. I'm using Etcher, and I copy a suitable wpa_supplicant.conf file to the root directory of the boot partition. But I do not have the same result as you. 

As soon as I disable FTP or Samba in the admin interface, the wifi configuration is lot a few seconds before my raspberry pi zero W reboots : actually, it seems that the key_mgmt key is modified in /data/etc/wpa_supplicant.conf file. It may be (but I'm not sure) the reason why I lose wifi. I had this behaviour dozens of times unfortunately yesterday. Easy to reproduce. 

I never got this 0.0.0.0 ip address. Yes I modified my S35Wifi file (and I do not have any more problem) but I also created two symbolic links from /data/etc/wpa_supplicant.conf to /boot and from /etc/wpa_supplicant.conf to /boot. 

Strange to have two different behaviours... 

Regards

JohBod, 2017-05-07:
I used the example from:

https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-%28Plus%29-And-Zero

to write a wpa_supplicant.conf file that looks like this:

**country=se
update_config=1
ctrl_interface=var/run/wpa_supplicant

network={
	scan_ssid=1
	ssid="blablablablabla"
	psk="blablablablabla"
}**

The lines eapol_version=, ap_scan=, fast_reauth=, proto= and key_mgmt= are not present in my file. Are you sure these lines are necessary in your case?

/J

VincentGaudeul, 2017-05-07:
My Wifi is WPA-PSK [TKIP] + WPA2-PSK [AES] encrypted. 
I followed your advice above. Here is my simplified wpa_supplicant.conf file : 

country=FR
update_config=1
ctrl_interface=var/run/wpa_supplicant

network={
	ssid="Beedoo24"
	psk=xxxxxxxxxx
	scan_ssid=1
}

I installed a new motioneyeos SD card in my raspberry pi zero W with that file. Wifi got up without any problem, but when I disabled FTP on admin interface, the wifi configuration was broken : 

* Starting wpa_supplicant: failed (brcmfmac)

Easy to reproduce I'm afraid.

ccrisan, 2017-05-07:
What does your wpa supplicant conf look like after you apply the ftp change? What exactly gets broken/changed?

VincentGaudeul, 2017-05-08:
Ok. Let's start from a fresh motioneyeos installation. Here are the three wpa_supplicant.conf files in the system. 

```
[root@meye-40fc089c ~]# cat /boot/wpa_supplicant.conf 
country=FR
update_config=1
ctrl_interface=var/run/wpa_supplicant

network={
	ssid="Beedoo24"
	psk=xxxxxxxx
	scan_ssid=1
}
```

```
[root@meye-40fc089c etc]# cat /etc/wpa_supplicant.conf 
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
}
```

```
[root@meye-40fc089c etc]# cat /data/etc/wpa_supplicant.conf 
country=FR
update_config=1
ctrl_interface=var/run/wpa_supplicant

network={
    ssid="Beedoo24"
	psk=xxxxxxxxxxx
	scan_ssid=1
    key_mgmt=NONE
}
```

Have you noticed that /data/etc/wpa_supplicant.conf file is already a modified version of /boot/wpa_supplicant.conf file ? That'is the first time I notice this... so I try to reboot my raspberry pi zero W, before any modification to any file on the system. Are we ok with this ? I just booted a fresh version of motioneyeos, with my wpa_supplicant.conf, I ssh-ed the raspberry, and typed "reboot". 
And then the wifi configuration is already broken : 

* Starting wpa_supplicant: failed (brcmfmac).

ccrisan, 2017-05-08:
Aren't you missing quotes around the psk value? The motionEye UI looks for them and in their absence considers that this is an open network, hence the key_mgmt=NONE.

VincentGaudeul, 2017-05-08:
psk needs quotes when psk represents the passphrase, which is not my case here. 

I used wpa_passphrase script to generate a 256-bit pre-shared WPA key from an ASCII passphrase and a given SSID. So my passphrase is never displayed anywhere. 
I guess motionEye UI is wrong if it looks for quotes to determine if this is an open network or not. That may be the point.

ccrisan, 2017-05-08:
Argh... why do all these WiFi configurations need to be so complicated. I'll take a look over the configuration file and possible values and see how this can be fixed.

Anyways, until then, given that your configuration is not a regular one, I'd recommend completely disabling motionEye's WiFi management support by commenting out the wpa_supplicant_conf line in `/data/etc/motioneye.conf`. Your wpa_supplicant.conf file should then be left intact whatever you change in the UI.

On a side note, `/boot/wpa_supplicant.conf`, if present, is used only once, at first boot. `/etc/wpa_supplicant.conf` is used in its absence and means no configured WiFi network. The `/data/etc/wpa_supplicant.conf` file is the one that is actually used and can be edited as per your needs.

I hope this helps.

VincentGaudeul, 2017-05-08:
Ok great. Yes wifi configurations are a mess :-) 
No hurry to fix the problem, I've already found a solution to this problem. 
And thanks for the support.

closed 2017-06-01 by bortek

VincentGaudeul, 2017-06-03:
This issue would some day require a fix...