motioneye-project/motioneyeos #932: Total memory showing as 117MB on Raspberry Pi 3

cuddylier, 2017-05-06:
Hi

I'm on the latest version, 0.35.2, and having a lot of issues with uploading to Dropbox and emails no longer working after I believe the last few updates. The web interface and SSH also became very slow at times even after a reboot.

I checked 'free -m' and 'top' and they both show I only have 117MB RAM on my Raspberry Pi 3.

 free -m
             total       used       free     shared    buffers     cached
Mem:           117         78         38          2          0          8
-/+ buffers/cache:         69         47
Swap:            0          0          0

I only have 128MB assigned to the GPU.

I did a quick Google and found people reporting the same issue with other Pis, seems to maybe be a kernel issue?

Thanks

ccrisan, 2017-05-07:
Is this motionEyeOS or Raspbian?

cuddylier, 2017-05-07:
Motioneyeos.

ccrisan, 2017-05-07:
@cuddylier what version? Don't say "the latest".

cuddylier, 2017-05-07:
Already provided in the OP but here is the 'OS version' too: motionEyeOS 20170329

ccrisan, 2017-05-07:
@cuddylier actually this is the first time you have disclosed the OS version. The other version you gave us was that of motionEye.

I couldn't find any reference online about people having their RPi3 reporting less than ~800MB. What do  `vcgencmd get_mem gpu` and `vcgencmd get_mem gpu` say?

cuddylier, 2017-05-07:
`vcgencmd get_mem gpu`

gpu=128M

What should the 2nd command be? You said the same command twice.

ccrisan, 2017-05-07:
My apologies. It's the same but with *arm* instead of *gpu*.

cuddylier, 2017-05-07:
@ccrisan 

`vcgencmd get_mem arm`
arm=128M

ccrisan, 2017-05-07:
Damn this beats me. What does cat/proc/cpuinfo say?

cuddylier, 2017-05-07:
@ccrisan https://pastebin.com/raw/MSq6tRct

ccrisan, 2017-05-07:
Have you tried Raspbian? Does it report the full amount of memory?

cuddylier, 2017-05-07:
I haven't no as I'd prefer not having to wipe the Pi (got no other micro SD cards) as I've had it installed with motioneyeos for a long time, this  only occurred in one of the few recent updates I applied, I usually check for updates every 1 - 2 weeks.

cuddylier, 2017-05-07:
The last day everything worked properly was the 29th April so an update on or before then.

closed 2017-06-01 by bortek