motioneye-project/motioneyeos #933: DNS Queries every 20 seconds and fifteen minutes

TomFaulkner, 2017-05-07:
I noticed Motion Eye OS on a Pi Zero W is quering www.google.com every twenty seconds and google.com every fifteen minutes.

I'm curious as the reason for this behavior.
[MotionEye OS google queries.pdf](https://github.com/ccrisan/motioneyeos/files/981894/MotionEye.OS.google.queries.pdf)

wdawson001, 2017-05-07:
From my understanding: One checks the date time the other checks that motioneyeos is connected to the internet. 

You can change this in the "expert settings" (once you've enabled advanced setting)

ccrisan, 2017-05-08:
As @wdawson001 said, if you enabled Connectivity Watch in Expert Settings, it will test the connectivity to www.google.com every 20s. This is configurable and disabled by default.

System date is updated every 15 minutes from google.com. This can also be configured.

closed 2017-06-01 by bortek