motioneye-project/motioneyeos #937: No Fast Network Camera

hplus28, 2017-05-12:
HI,
I am using a raspi zero (both W and without built in wifi) and a csi camera. 
I don't see the dast network camera option in the expert settings.
Expert Settings is only showing Log Files et Enable system monitoring. Is this normal?

Thanks

ccrisan, 2017-05-12:
Screenshot please.

hplus28, 2017-05-12:
[image]

ccrisan, 2017-05-12:
This is weird. Can you do a clear cache and a full refresh? If the issue persists, please attach your log files.

hplus28, 2017-05-12:
I also don't see any network configuration in the interface.
I had to first boot with a screen and keyborad plugged in and manually edit the wpa_supplicant.conf in /data/etc in order to connect

ccrisan, 2017-05-12:
You may need to rewrite your OS from scratch as things aren't alright there with your configuration.

hplus28, 2017-05-12:
yeah... my only problem is thas i don't have a linux to use the writeimage.sh to set the ssid, that why i did it by hand

ccrisan, 2017-05-12:
You don't need a Linux. Use your windows method to write the image and then use your windows to put the wpa_supplicant.conf file on the boot partition (which is a fat32). Just make sure to do this before the first boot.

hplus28, 2017-05-12:
you mean i can put the correct wpa_supplicant.conf in the visible partition under windows before the first boot? i'll try this

ccrisan, 2017-05-12:
Sure. See https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-%28Plus%29-And-Zero.

hplus28, 2017-05-12:
I've just tried this using the same wpa_supplicant.conf that was on the raspi zero and i've copy it on a newly flashed sd card with the latested image. It keeps on rebooting trying to configure wired network and saying no device found....
The wpa_supplicant.conf is correctly written... no mistake its a real .conf file (not a .conf.txt )

hplus28, 2017-05-12:
ok i made it work, i've first booted on a rasp B, it connected thru the wired network, then using the web interface, i first enabled the wireless network and configured the ssid and password.
I finally got all the options...
I think that something is wrong in the image, maybe wireless is not enabled by default?

closed 2017-06-01 by bortek