motioneye-project/motioneyeos #939: RPi Zero W won't upload to Google Drive

cephasiii, 2017-05-13:
I have three RPi's: (1) RPi 3 running motioneye ontop of raspberian, (2) RPi Zero W running motioneyeos. The RPi 3 and one of the RPi Zero W upload to my Google Drive account flawlessley. However, the other RPi W does not upload. I have reinstalled motioneyeos a couple of times, requested a new authorization key and ensured the settings for both RPi W were the same but no luck.
[TOP RPi Zero W.txt](https://github.com/ccrisan/motioneyeos/files/998261/TOP.RPi.Zero.W.txt)
[motioneye.2.txt](https://github.com/ccrisan/motioneyeos/files/998262/motioneye.2.txt)

Any suggestions? Thanks in advance for your support.

silversword411, 2017-05-17:
I was reading thru another issue, and there was this article: https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-(Plus)-And-Zero

The Zero W apparently just barely has enough RAM for the standard motioneyeos components. You're probably running out of RAM.

cephasiii, 2017-05-18:
@silverwood411, thank you for the response. As I mentioned in my initial post, I have two RPi Zero W's running motioneyeos. One of the two is not uploading pictures/video to my Google Drive account. Not sure why one would work and the other one wouldn't...

silversword411, 2017-05-18:
I think this means out of memory: 
Starting at Line 154
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 659, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 94, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 202, in upload_data
MemoryError

Had the same problem with power adapter on a Pi 3. Work for several days, then intermittently fail. When power usage spiked (due to enough events at once) it'd drop off wifi, then have to reboot it to get it going again.

Swapping the MicroSD card between devices, then re-test see if the problem moves with the card. 
Also disable all the unneeded services possible (ssh, ftp, smb etc).

Not sure why linux memory manager doesn't swap out temporarily, slow the system...but at least keep it working.

silversword411, 2017-05-18:
You're also getting movie file too large back from gdrive, looks like they're capping you at 1024MB, you're getting 1400MB+ files (that's video files of 1GB you realize?):

2017-04-23 12:00:51: [motioneye]    ERROR: gdrive: file "/data/media/sda1/2017-04-23/11-30-42.mp4" is too large (2353MB/1024MB)
2017-04-23 12:00:51: [motioneye]    ERROR: failed to upload file "/data/media/sda1/2017-04-23/11-30-42.mp4" with service gdrive: file "/data/media/sda1/2017-04-23/11-30-42.mp4" is too large (2353MB/1024MB)

You check your space available in your account?

Also there's rate limiters build into the API...you start hammering it with too much data, it'll throttle you back.

closed 2017-05-18 by silversword411