motioneye-project/motioneyeos #940: automatically save timelapse video daily

Jasonb11, 2017-05-13:
I want to download timelapse videos daily to a nas box on my network. 
I am using a raspberry pi for motioneyeos and gargoyle router software. 

What is the best way to do this? 
Cron?
Python?
Something else?

I'm open to suggestion. 
Thanks.

Chiny91, 2017-05-13:
Bit of an open question really as there must be many ways to achieve that end.   What do you prefer; what are you comfortable with ?

cron is simply the timekeeper; you would need something to actually "download"... you could use something simple like cp (Linux copy) but this is not secure.   If you are slightly paranoid (a good thing), then perhaps scp would be better.   Inevitably, with greater security comes greater inconvenience.

Or maybe your NAS box comes with suitable s/w to slurp files from motionEyeOS.   Many NASes do such things; obviously it depends on what you have and what you are prepared to hack.

Personally, I  use rsync to transfer photo stills (rather than make the humble cpu in my R Pi work harder in producing videos) under cron.   rsync comes with some useful config options, such as the ability to limit the bandwidth in the transfer, which I find useful as the R Pi is restricted in its ethernet capability.   ISTR that I nice/ionice the cron job to unknown real effect, just in case, so as to keep max R Pi oomph available for motion.   It has worked for a long time, without hassle, although I'm not claiming it is the "best way" 😃

Jasonb11, 2017-05-16:
I am using gargoyle router software which is based on opwnwrt. 

I am confortable with whatever. I looked up the ffmpeg command but i dont really get how i can create the timelapse automatically by writing the script myself and there aren't too many examples of how it is done on a daily basis and from a nas location.

Chiny91, 2017-05-16:
OK, well perhaps you could run everything on your motionEyeOS R Pi.

(a) The automation is by cron as you suspect, as per the [Wiki](https://github.com/ccrisan/motioneyeos/wiki/Tweaks).
(b) The ffmpeg command, as given by @ccrisan in 2015 could be:
`cat *.jpg | ffmpeg -framerate 5 -f image2pipe -vcodec mjpeg -i - -vcodec mpeg4 -b:v 9999999 -qscale:v 0.1 -f avi my_timelapse.avi`
That worked in 2015 but I've not tested it since.

So, you would make a crontab (crontab -e) and insert the tested command after the timing.   I usually use a text editor to assemble the command, copy/paste it to a terminal to test, copy/paste it to crontab -e finally.    Check your /var/log/messages during each process (assemble, test, cron) for problems.

If you have a low-power R Pi, with hi-res camera, your R Pi might glitch with all the effort at times

Jasonb11, 2017-05-16:
How will this this command work on the NAS location? Is there more to the script to open the correct directory everyday? How could I make the my_timelapse.avi file have sequential numbers to keep everything straight?

Jasonb11, 2017-05-18:
I figured it out:

*First*
`crontab-e`

*Then*
 `58 23 * * * cd /home/ftp/storage/my-NAS-location && cd  "$(\ls -1dt ./*/ | head -n 1)"  &&  cat *.jpg | ffmpeg -framerate 25 -f image2pipe -vcodec mjpeg -i - -vcodec mpeg4 -b:v 9999999 -qscale$`

*Finally*
save and exit

I added date and time to the filename so there isnt any naming issues. Can't wait to see what a whole year of timelapse videos will look like.

Jasonb11, 2017-05-18:
Thank you @Chiny91 for helping me get started to figuring this out!

closed 2017-05-18 by Jasonb11

sfgabe, 2018-02-12:
You can also do this outside of motioneyeos with imageio (I have it running on my server but I suppose you could run it anywhere as a chron). This is just another option if you want more control over the frame rate, number of photos to include, and other movie parameters.

    import os
    import time
    import datetime
    import glob
    import imageio
    imageio.plugins.ffmpeg.download()
    from collections import namedtuple
    
    ImageFile = namedtuple('ImageFile', ['img', 'filename'])
    
    today = datetime.date.today().strftime("%Y-%m-%d")
    todayLong = datetime.date.today().strftime("%B %d, %Y")
    now = datetime.datetime.now().strftime("%I%M%m%d%Y")
    cwd = os.getcwd()
    dir_path = os.path.dirname(os.path.realpath(__file__))
    
    imgURL = '.path/to/images/'
    gifPath = os.path.join(imgURL, today)
    gifOutputPath = os.path.join('output/octo_gif_' + now + '.mp4')
    gifOutputName = os.path.join('octo_gif_' + now + '.mp4')
    fullgifPath = os.path.join(gifPath, '*.jpg')
    list_of_files = glob.glob(fullgifPath)
    latest_file = max(list_of_files, key=os.path.getctime)
    
    def get_images():
    images = []
    for subdir, dirs, files in os.walk(gifPath):
        for file in sorted(files)[-40:]:
            file_path = os.path.join(subdir, file)
            if file_path.endswith(".jpg"):
                images.append(imageio.imread(file_path))
                kargs = { 'fps': 3, 'quality': 10, 'ffmpeg_params': ['-s','640x400','-probesize','20M'] }
                imageio.mimsave(gifOutputPath, images, 'FFMPEG', **kargs)

After this you can view todays movie through the file system, have it emailed to you or get posted somewhere else. You can see an example of the results on twitter at [Octo_Cam_Bot](https://twitter.com/Octo_Cam_Bot)

afeno, 2018-10-31:
Look at a simple way to create a timeleapse file using the "Extra Motion Options" directly from MotionEyeOS:
https://github.com/ccrisan/motioneyeos/issues/861