motioneye-project/motioneyeos #941: Web server with PHP?

schmeckles22, 2017-05-13:
Hello,

I love MotionEyeOS and I was wondering if I could host a web server and PHP on the machine too, I have never installed without apt-get,yum or pacman before and don't want to be compiling from source to only cause more issues as I only have remote access right now.

The reason being is I have made a PHP script to check if the user is on a the same IP as the server so redirect them to the local LAN IP and if they're external then take them to the WAN IP address.

ccrisan, 2017-05-13:
You normally can't install anything like that on motionEyeOS. I'd recommend using regular linux distros for that (such as Raspbian). If you really need PHP on motionEyeOS, I'd recommend forking it, familiarizing with BuildRoot and recompiling everything.

schmeckles22, 2017-05-13:
@ccrisan Thank you, I originally started with a Chinese Mini-PC with an Intel Atom processor running Arch Linux + MotionEye but it didn't want to accept the Networked IP Camera. If I install Raspbian will MotionEye work the same as MotionEyeOS? Because at the minute everything works spot on.

ccrisan, 2017-05-13:
Most of the things should work just as well as they work on motionEyeOS. Just make sure to install motion version 4.x.

closed 2017-05-13 by schmeckles22