motioneye-project/motioneyeos #942: SMS notification help??? HOw

galelong, 2017-05-13:
How to do a SMS notification in motioneyeos only email notification is the option. i search here but someone said that i can be done throught webhook ?? if it can can someone please help me do this ?. no pm here so please help me this is my email galelong07@gmail.com

Ccrisan i also see this https://www.youtube.com/watch?v=smqB-YBGP_U tutorial about sending SMS but where can i code in motioneyeos can i install a module ? like in the said video. The video use twilio to send SMS notification thank you so much for the help...

keithellis74, 2017-05-13:
Your can use the webhook with the ifttt.com service.  If you have iOS, you can also set up iOS notifications.

Signup to ifttt.com and create a new applet.  Search for "maker" and create a web hook, select "receive web request" as trigger, and give it a name.  As the "action" select "SMS" or "notification"

Then go to "my services" search for "maker" and there is a URL, add this URL into the webhook section in MotionEyeOs

galelong, 2017-05-13:
hi Keithellis74 i did what you say ? but i paste the url in my account info ?? but can you guide how will that url send an SMS notification to my Phone number ??? thank you...

rynerlute159, 2017-05-14:
I need this too for my shool project please help me T_T.. I don't know how to send SMS in python using twilio in webhook i tried ifttt but no luck. i also try coding with php i set up my webhosting then upload my send-sms.php then call it sendsms.py . Bump for this i also really need this one.

closed 2017-06-01 by bortek