motioneye-project/motioneyeos #945: Cam doesen´t work any more after update to Version 0.35.2 (newest)

Oberlandix, 2017-05-16:
Dear Calin, dear motionEYEos-Friends,

yesterday I did the update. Since this day, the "software works" but there is no picture anymore. The Cam (Raspi-Cam on Board) doesen´t work any more. Reboots done. Any ideas?

https://picload.org/view/rirrcrlw/cam_prob3.png.html

https://picload.org/view/rirrcrcr/cam_prob2.png.html

https://picload.org/view/rirrcrca/cam_prob1.png.html

Thank you for your help -> In case of emergency I must reinstall the image... But the Raspi is built in very well... :-/

Oberlandix

ccrisan, 2017-05-16:
Please attach your motion.log file.

Oberlandix, 2017-05-16:
Hi Crisan, thank you very much for your quick reply.
The motion.log:
[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1+git37b3595 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

ccrisan, 2017-05-16:
Hmm motion doesn't seem to complain about anything. Can you attach your motioneye.log file?

Oberlandix, 2017-05-17:
Sure! Thank you very very much!

Oberlandix, 2017-05-20:
Any news Calin? Did you receive my email with the motioneye.log file? Greetz

ccrisan, 2017-05-20:
Nope. Please don't use email replies when commenting on github, as attachments are ignored.

Oberlandix, 2017-05-22:
Good morning,
the problem is, that I can´t get the motioneye.log file TODAY anymore :-( The "link is a link" but it is no file generated.

The old one is this:

2016-08-21 10:47:16: [motioneye]     INFO: hello! this is motionEye server 20160410
2016-08-21 10:47:16: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2016-08-21 10:47:16: [motioneye]     INFO: cleanup started
2016-08-21 10:47:16: [motioneye]     INFO: wsswitch started
2016-08-21 10:47:16: [motioneye]     INFO: tasks started
2016-08-21 10:47:16: [motioneye]     INFO: mjpg client garbage collector started
2016-08-21 10:47:16: [motioneye]     INFO: smb mounts started
2016-08-21 10:47:16: [motioneye]     INFO: server started
2016-08-21 10:47:17: [motioneye]     INFO: adding new camera with id 1...
2016-08-21 10:47:17: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2016-08-21 10:50:32: [motioneye]    ERROR: authentication failed for user biene
2016-08-21 10:56:01: [motioneye]    ERROR: authentication failed for user Biene
2016-08-21 10:59:38: [motioneye]    ERROR: authentication failed for user Biene
2016-08-21 11:00:14: [motioneye]    ERROR: authentication failed for user admin
2016-08-21 11:00:25: [motioneye]    ERROR: authentication failed for user biene
2016-08-21 11:00:39: [motioneye]    ERROR: authentication failed for user Biene
2016-08-21 11:02:40: [motioneye]    ERROR: authentication failed for user Admin
2016-08-21 11:10:28: [motioneye]    ERROR: authentication failed for user admin
2016-08-21 11:16:30: [motioneye]     INFO: rebooting
2016-08-21 11:16:30: [motioneye]     INFO: executing "/sbin/reboot"
2016-08-21 11:16:31: [motioneye]     INFO: interrupt signal received, shutting down...
2016-08-21 11:16:31: [motioneye]     INFO: server stopped
2016-08-21 11:16:31: [motioneye]     INFO: tasks stopped
2016-08-21 11:16:34: [motioneye]     INFO: motion stopped
2016-08-21 11:16:34: [motioneye]     INFO: smb mounts stopped
2016-08-21 11:16:34: [motioneye]     INFO: bye!
2016-08-21 11:16:48: [motioneye]     INFO: hello! this is motionEye server 20160410
2016-08-21 11:16:50: [motioneye]     INFO: cleanup started
2016-08-21 11:16:50: [motioneye]     INFO: wsswitch started
2016-08-21 11:16:50: [motioneye]     INFO: tasks started
2016-08-21 11:16:50: [motioneye]     INFO: mjpg client garbage collector started
2016-08-21 11:16:50: [motioneye]     INFO: smb mounts started
2016-08-21 11:16:51: [motioneye]     INFO: server started
2016-08-21 11:17:49: [motioneye]     INFO: updating to version 20160705...
2016-08-21 11:17:49: [motioneye]     INFO: killing motioneye init script...
2016-08-21 11:17:49: [motioneye]     INFO: stopping netwatch init script...
2016-08-21 11:17:49: [motioneye]     INFO: downloading https://github.com/ccrisan/motioneyeos/releases/download/20160705/motioneyeos-raspberrypi3-20160705.img.gz...
2016-08-21 11:20:55: [motioneye]     INFO: decompressing /data/.firmware_update/firmware.gz...
2016-08-21 11:21:11: [motioneye]     INFO: reading partiton table...
2016-08-21 11:21:17: [motioneye]     INFO: extracting boot.img...
2016-08-21 11:21:17: [motioneye]     INFO: extracting root.img...
2016-08-21 11:21:31: [motioneye]     INFO: backing up /boot/config.txt
2016-08-21 11:21:31: [motioneye]     INFO: unmounting boot partition...
2016-08-21 11:21:31: [motioneye]     INFO: installing boot image...
2016-08-21 11:21:41: [motioneye]     INFO: mounting boot partition read-write...
2016-08-21 11:21:46: [motioneye]     INFO: restoring up /boot/config.txt
2016-08-21 11:21:46: [motioneye]     INFO: preparing to boot in fwupdate mode...
2016-08-21 11:21:46: [motioneye]     INFO: rebooting...
2016-08-21 11:21:51: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2016-08-21 11:21:52: [motioneye]     INFO: interrupt signal received, shutting down...
2016-08-21 11:21:56: [motioneye]     INFO: server stopped
2016-08-21 11:21:56: [motioneye]     INFO: tasks stopped
2016-08-21 11:22:57: [motioneye]     INFO: hello! this is motionEye server 20160705
2016-08-21 11:22:59: [motioneye]     INFO: cleanup started
2016-08-21 11:22:59: [motioneye]     INFO: wsswitch started
2016-08-21 11:22:59: [motioneye]     INFO: tasks started
2016-08-21 11:22:59: [motioneye]     INFO: mjpg client garbage collector started
2016-08-21 11:22:59: [motioneye]     INFO: smb mounts started
2016-08-21 11:22:59: [motioneye]     INFO: server started
2016-08-21 11:24:21: [motioneye]  WARNING: 403 GET /login/?_=1471778645603&_username=admin&_signature=204fa3123d6848a379d1db40cbbf5a33fb864981 (192.168.2.51) 2.44ms
2016-08-21 11:25:03: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=ffb20556f847a7efb6e55124713cf6b91ceaaeb6 (127.0.0.1) 3.20ms
2016-08-21 11:25:03: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=ffb20556f847a7efb6e55124713cf6b91ceaaeb6 (127.0.0.1) 3.16ms
2016-08-21 11:28:00: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=ffb20556f847a7efb6e55124713cf6b91ceaaeb6 (127.0.0.1) 5.93ms
2016-08-21 11:28:00: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=ffb20556f847a7efb6e55124713cf6b91ceaaeb6 (127.0.0.1) 5.40ms
[... cut ...]
2016-08-28 20:10:00: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=28ad2415b6d4aa75b8100872224cadf9e94d1f59 (127.0.0.1) 2.89ms
2016-08-28 20:15:00: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=c5a33c322743bf5bbc3c8043666cf66357aec025 (127.0.0.1) 2.93ms
2016-08-28 20:20:00: [motioneye]  WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=9b94e89fa503cdbbd003fbeee6b13bcb21b2ee77 (127.0.0.1) 2.98ms
2016-08-28 20:25:00: [motioneye]  WARNING: 403 POST /_relay_event/?_username=xxxxxx

Thank you very much, Oberlandix

Oberlandix, 2017-05-29:
OK... I think I must do a full new installation...?! Greetings, Oberlandix

closed 2017-06-01 by bortek