motioneye-project/motioneyeos #948: Can't add new camera on motioneye

topdiogo, 2017-05-16:
Hello, i just got some ip cameras 
[https://www.aliexpress.com/item/2MP-Full-HD-1080p-PoE-Camera-IP-Security-NOVIF-2-0-SONY-CMOS-IR-Night-Vision/32553616394.html?spm=2114.13010608.0.0.Ma4lrT](url)

and a raspberry 3 model b, so i can set up a cctv system.

Can you tell me why i'm not able to add any of the cameras to the software?
Camera type: Network Camera
URL: http://192.168.1.201:8081/
Username:
Password:
Always get a "request timeout error"
I got the stream working on browser... 
Could you help me please?

silversword411, 2017-05-17:
You have to have a camera that will output video using one of these formats for compatibility with motioneyeos: RTSP, MJPEG or JPEG URL

You have to determine what the URL is for that camera for the stream in question. You'll have to contact camera developer to determine what those URL's are if you can find it in the settings of the camera.

topdiogo, 2017-05-17:
So there isn't  any chance to use a h264 cam on motioneye? 
There isn't an url stream for these cameras as well?

silversword411, 2017-05-17:
As far as I'm aware, these are the only network based camera feeds supported by motioneyeos at this time.
https://i.imgur.com/9DjYkQi.png

closed 2017-05-17 by silversword411