motioneye-project/motioneyeos #949: Wifi configuration during install from Windows (was needing for Pi Zero W)

silversword411, 2017-05-17:
Was going to try and improve Installation notes:
https://github.com/ccrisan/motioneyeos/wiki/Installation
so that injecting the wifi SSID/pass can be done from windows.

In another Pi Zero W thread:
https://github.com/ccrisan/motioneyeos/issues/804

There was a reference to: 
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting-Raspberry-PI-Models-A-(Plus)-And-Zero

If this is applicable to any motioneyeos install, I'll update the installation page:
---------------
Prepare a wpa_supplicant.conf file and place it on the boot partition of your SD card; here's what the file should look like:

 update_config=1
 ctrl_interface=/var/run/wpa_supplicant

 network={
     scan_ssid=1
     ssid="your_network"
     psk="your_password"
 }
--------------

jasaw, 2017-07-18:
@silversword411 What about adding `country=XX` at the top of the file?

Adding a list of country codes would be very helpful too.

silversword411, 2017-07-20:
I still haven't figured out HOW to even use the file/inject the wifi into a Pi Zero W. I never heard from anyone that knows how the file works. I'm just looking for information so I can improve the documentation.

If you want to enhance what the wpa_supplicant file can do, I'm not sure where to start with that.

jasaw, 2017-07-20:
@silversword411 Checkout issue #1010 latest reply for steps on how to configure Wifi during install from Windows. It applies to Pi Zero W.

closed 2017-07-21 by silversword411

jasaw, 2017-07-21:
@silversword411 It may be very helpful to others if you could update the documentation.

silversword411, 2017-07-21:
I was working on it at the time of writing. ;) I hate wiki markup...why isn't there a WYSIWYG editor yet :/