motioneye-project/motioneyeos #953: Night Sensitivity

dgilbert2, 2017-05-20:
I am using motioneye with the NOIR V2 camera but at night the motion images are virtually solid black.

I have been trying various settings and adjustments to increase the cameras sensitivity / iso at night without luck.

How can I add "Extra Motion options" or motion.conf changes to resolve this please?

wdawson001, 2017-05-20:
The noir camera doesn't filter our IR light. 

If you want the camera to see at night, you need to add an IR light source.

dgilbert2, 2017-05-20:
Thanks wdawson100, yes I understand that, the default sensitivity though is not that high, so I'm wanting to increases it to help with low light / IR light also. I've seen it can be done but cannot see how to do it with motioneye.

dgilbert2, 2017-05-21:
Following further research I see that the latest Motion version allows us of mmalcam commands. This would allow a command to control the camera exposure as I understand it, eg;
mmalcam_control_params -ex sports -ev 25
or
mmalcam_control_params -ex night

Therefore in motioneye's Extra Motion Options I have added;

mmalcam_name vc.ril.camera
mmalcam_control_params -ex night

These are appearing in the thread-1.conf file but do not seem to change the camera settings.

Therefore, is the camera name vc.ril.camera correct when using motioneye and do indeed mmlcam commands work?

Thanks for any help.

CaelynInc, 2017-06-03:
I have basically the same question. For example, night performance without extra ir leds when using rpi web interface set to nightpreview or night mode, is a lot better...

dgilbert2, 2017-06-03:
From what I have tried it cannot be done :-(

However, under Expert Settings you can select Fast Network Camera which then allows the ISO settings etc to be configured. BUT, this disables recording and motion detection so you then have to use something like AngelCam who are developing a motion detection system.

CaelynInc, 2017-06-03:
That's the thing... I'd really like to be able to use the motion detection functionality. Insee how the difference exists though, in Fast Network Camera mode, the camera runs as an mjpeg stream device (I believe from what I found because it runs via the gpu in this mode) whereas in regular mode it runs as an CPU based v4l2 device. I am not that familiar with the capabilities of the v4l2 devices and settings, but you'd think something like it should be possible, right? It's all about setting the exposure as far as I know...

I can see where on a pi zero w in my case, it might be that the CPU is not strong enough... But I'd at least like to know that ;-)

ccrisan, 2017-06-03:
One could use the `v4l2-ctl` command to adjust some of the camera settings, I believe. This doesn't need to be part of motionEye's UI.

dsoundzmedia, 2017-06-16:
Hello. I'm about to try your software/app with a Nor IR cam (I also have a Logitech c920 webcam). I have epilepsy, and I have seizures in the night, so a RPi and your software was something I've been looking for, as other epilepsy sleeping/night camera recording solutions ( such as the SAMi https://www.samialert.com/ ) cost too much, so I thought a DIY approach would be better. I will soon see what output and results I get with your software. I'll be able to add comments to this thread since I'll be using this at night. thanks @CaelynInc  for development.

fredericcollin, 2017-06-23:
dgilbert2: I use a 2nd RPI with also MotioneyeOS installed and capture the fast network camera stream from the 1st RPI. That way you can use recording & motion detection.

freddy64, 2017-06-30:
Hello, this is my very first post and I am new to the Raspbery Pi and Linux operating systems, so please bear with me. I have three rpi zero w and v2. 1 camera boards, one of which is the NoIr. I want to use these as security cameras. Up until now I've been streaming video from the raspivid command to vlc player with the Raspbian Jessie OS. I have just installed motioneyeOS onto a fresh sd card and I am very impressed, this is exactly what i wanted.
Is it possible to include the "-ex nightpreview" camera setting option that raspivd provides into motioneyeOS and if so, could motion still be detected ?
I have used this option with the NoIr camera board at night under poor light conditions, where a street lamp in front of my house is the only light source and it works well.

chappy1978, 2017-07-10:
@ccrisan I really feel like there should be a build in command for low light. I played around with v4l2-ctl and had very little real luck with changing these settings. A "-ex nightpreview" camera setting option" would really be a bonus.

jasaw, 2017-09-01:
To play with night sensitivity, you'll need to use the MMAL camera interface, which means recompiling motionEyeOS with MMAL enabled.
Once you have MMAL compiled in, manually edit the thread-1.conf file.

Some examples I found online.

Example 1:
```
mmalcam_name vc.ril.camera
mmalcam_control_params -ex night
```

Example 2:
```
mmalcam_control_params -ex sports -ev 25 --shutter 256000
auto_brightness on
brightness 255
contrast 1
saturation 5
hue 0
```

ccrisan, 2017-09-01:
Guys, does this mean that the v4l2-ctl interface is broken or does not do the same thing as the MMAL one does? I've been meaning to implement a more generalized v4l2 user interface in motionEye so that all these controls are covered by it in a friendly way.

MiniSumo, 2017-09-18:
I do need to add an IR source for night but have noted that I can get better twilight images when I tweak the brightness/contrast manually - I will need to swap back in daylight.

I do hope that this appears in controls, I've still not got my head around v4l2-ctl and automating changes.

Alpengroove, 2017-12-31:
Any news on the „-ex night“ mode?

It would be so great to be able to use the parameter with motioneye. I run a webcam under jessie and I use raspistill with a simple script. The -ex night parameter lets me see the camera picture the through the whole night, with motioneye it is dark.

ccrisan, 2017-12-31:
@Alpengroove actually yes, a new prerelease is currently baking (can be built from the buildroot-next branch) and will support motion mmal parameters, including `-ex night`.

KarlBaumann, 2018-03-20:
@ccrisan How far did you get with this option? 
I had success with enabling night mode by adding 
`mmalcam_control_params -ex night`
to  /etc/motioneye/motion.conf

ccrisan, 2018-03-20:
@KarlBaumann 
1. `/etc/motioneye/motion.conf` is not present on motionEyeOS so I assume you're on motionEye.
2. You should normally use the *Extra Motion Options* field in the settings panel to add, well, extra motion options. Alternatively you could manually add them to `/etc/motioneye/thread-1.conf`.

KarlBaumann, 2018-03-20:
Yes, sorry, I am always mixing those projects up.
I tried to use _Extra Motion Options_ but had no success.
What should I write there to achieve the same result as _mmalcam_control_params -ex night_
in /etc/motioneye/motion.conf ?

GuLinux, 2018-03-20:
I'm currently using motioneye as a night camera, I do use IR leds, but I still found out that brightness needed to be increased.

I wrote a python application that basically wraps [streameye](https://github.com/ccrisan/streameye) (written by @ccrisan as well), and uses pyephem to calculate sun position (positive, meaning above horizon, or negative, below) and then relaunch streameye with the proper parameters.

An example night setting would be 
```
/home/pi/streameye/extras/raspimjpeg.py -w 1600 -h 1200 -r 1.65 -q 20 --iso 800 --shutter 600000 --exposure night --awb sunlight
```
The shutter and `-r` parameters are very important, as increasing exposure will also help gathering more light.

With these settings I can easily see my whole garden (and it's quite a big one).
You also need of course to be able to relaunch streameye with different parameters, in order to be able to capture daylight pictures as well (with this settings it will be extremely overexposed).

After I'll to polish the code a little bit and add a configuration file, I might push this into a repo, or just create a PR to add it as an extra to `motioneye` or `streameye` repos.

ccrisan, 2018-03-20:
@KarlBaumann place that line in `/etc/motioneye/thread-1.conf`, not in `/etc/motioneye/motion.conf`.

alex-mex, 2019-07-16:
have a similar issue with NOIR V2 camera. When it gets dark the camera does not adjust unless i change any setting and then change it back so that "Apply" button appears. After clicking the "Apply" button the brightness adjusts to desirable level. I tried to run cronetab  "0 1 * * * killalll motion; sleep 8; motion /data/etc/motion.conf;"  but the image hangs.    I was wondering if anyone knows what is the execution code behind "Apply" button is, so that i may try to execute that in cronetab instead.  Basicaly need to execute code behind "Apply" button during the hours when it gets dark and when it gets light. Otherwise you get black and white screens.

alex-mex, 2019-07-17:
seems like  15 sec delay does the trick. That is what they have in their js 15 sec delay.
EDITOR=nano crontab -e
0 * * * * killall motion; sleep 15; motion /data/etc/motion.conf;
Ctrl+o
Crtl+x
will restart motion at the start of every hour.