motioneye-project/motioneyeos #954: Trying to set up 2 PiZeroW fast cams with Pi3 MotionEyeOS server...

midinotes, 2017-05-20:
I have set up two separate PiZeroW MotionEyeOS cameras, one with a 5MP camera and one with a 8MP NOIR camera. Both work great on their own IP addresses and with Fast Network Camera mode enabled I am getting decent frame rates plus the ability to tweak the ISO for better night vision.
Now whilst I can enable motion detection and storage for both individually, this means dropping the ability to change ISO and it slows things down, even with the ZeroWs overclocked.
After looking at the usage scenarios, I would like to try and use a Pi 3 as a motioneye OS server that (I assume) could take the feeds from both of the fast network cameras and then use the Pi3's faster CPU to do the heavy lifting and motion detection etc?   
My first issue is that I can't add the ZeroW cameras as fast network cameras to MotioEyeOS running on the Pi3, only as non fast net cameras.  I can add them as Network cameras (not Motion Eye) but the performance isn't great. Am I doing this right?   If I turn off fast net cam mode on the ZeroW cameras running MotionEyeOS, I can add them as remote MotionEye cameras but then it seems the Pi3 is just providing a view of both cameras on one IP address and nothing else? Or is that how it works?
I did try to install MotionEye (not OS) on Raspbian but the installation appeared to fail in a few places with files not being found.  However if that is the direction I should go then I'll persevere and give it another go.

Thanks in advance for any advice/help and just want to say how much fun it is to use the Pi computers for building a camera network thanks to the great work of Calin Crisan!

ccrisan, 2017-05-20:
Things aren't this complicated :) I recommend that you keep your Zero as in Fast Network Camera mode and add it as a Network Camera to your RPi3 running as a regular motionEye(OS). That's the best you'll get out of your hardware in terms of performance.

midinotes, 2017-05-20:
Thanks for the reassurance Calin.  Problem is how should I add the fast network camera to the Pi3 running motioneyeOS?  It won't detect it even with the IP address:port plus admin details... ?  I end up having to add it as a network camera, yet if the Zero isn't in fast mode it will add?

Simon

ccrisan, 2017-05-20:
motionEyeOS is fine on the Pi3, but it could just as well be a regular Linux + motionEye.

What do you mean by "it won't detect it"? You *do* have to add it as a *Network Camera*, since your Zero is now a bare *Network Camera*. Unless you have tweaked stuff on the Zero, the following URL should work for you `http://zero_ip:8081/`.

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

closed 2017-06-01 by bortek

HeAnVu75, 2019-06-10:
Hi, bortek and Chris. I have issues with activating action buttons (light_on_1-4 and light_off_1-4) remotely to the fast network camera from my rsp pi 3B+ server. I have made the action buttons on all of the remote fast network cameras as (light_on_1* an light_off_1*) and I have made them on the server as light_on_1* till 4* and light_off_1* till 4*. Is there a way to use symbolic link to activate light_on_1* buttons and off on the fast network cameras or is there another way to remotely activate the action buttons? Or can I use SSH Client on MotionEyeOs with ssh-keygen -f /data/etc/ssh_id_rsa? I did not understand how to SHH. Can anyone give me the right support to activating external Action buttons and GPIO pins?

Regards HeAnVu75