motioneye-project/motioneyeos #955: update to the latest version of motioneyeos now reboots with rainbow screen

LanceSaintLuke, 2017-05-20:
Hello,

I recently clicked the upgrade button within motioneyeos and after the upgrade the OS would not boot up any longer.  I notice that the ethernet activity lights don't flash any longer.  If I plug a monitor into it all I see is big square pinkish color flash on the screen and then it goes back to black.  It keeps cycling back and forth between these screens as if it's rebooting.

Any ideas?

I looked at the microSD card on my macbook pro.

cmdline.txt shows this:
dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1 ipv6.disable=1

config.txt shows this:
start_file=start.elf
fixup_file=fixup.elf
arm_freq=900
core_freq=250
sdram_freq=450
over_voltage=0
gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128
disable_camera_led=1
gpu_freq=250
initramfs fwupdater.gz

Thank You,
Lance

ccrisan, 2017-05-20:
You might need to rewrite the os image from scratch, as things appear to have gone bad on your sd card.

LanceSaintLuke, 2017-05-20:
Thanks for the reply.  I was thinking that may be the only solution as well so I just pulled down the latest image and flashed a new sd card.

I have re-configured it best as I can remember.  (i've learned my lesson by the way and have already backed up the config).

Do you happen to remember how to disable the LED light on the Pi camera?

LanceSaintLuke, 2017-05-20:
Wasn't there a button to disable the camera led in the expert settings section in the previous releases?

bortek, 2017-06-01:
read the wiki about turning off the LED. rainbow icon means you have power issues.

closed 2017-06-01 by bortek