motioneye-project/motioneyeos #960: Firewall "iptables" configuration

Laksuu, 2017-05-23:
Hi. I just downloaded your newest version of motioneyeos (motioneyeos-raspberrypi3-20170329.img.gz) on Raspberry PI3B and everything is fine except the firewall.

I found out that motioneyeos has iptables as firewall and i´m trying to configure it. I successfully configured iptables and saved the configuration but after saving it, it will automatically reboot and all iptables configurations are gone. It keeps doing that everytime i try to save iptable configurations.

is this a bug or am i doing something stupid?

Commands what i have been using in iptables configuration...

iptables -L

iptables -s 192.168.0.20 -A INPUT -j ACCEPT
iptables -A INPUT -p tcp --dport 8081 -j ACCEPT

iptables -A INPUT -p TCP -m state --state RELATED,ESTABLISHED -j ACCEPT
iptables -A INPUT -p UDP -m state --state RELATED,ESTABLISHED -j ACCEPT

iptables -P INPUT DROP
iptables -P FORWARD DROP
iptables -P OUTPUT ACCEPT

iptables-save

ccrisan, 2017-05-23:
`iptables` is by it's nature non-persistent. You're supposed to run all those commands from a startup script, normally `/data/etc/userinit.sh`).

`iptables-save` is just a tool that dumps the firewall configuration to output, not saving anything actually. You could use it in conjunction with `iptables-restore` and shell redirects but it's harder than executing your commands at startup.

closed 2017-05-23 by ccrisan

Laksuu, 2017-05-23:
Alright. I try that. Thanks for quick a answer!

And btw. Awesome work on Motioneyeos! GUI and everything is running like a dream and looks cool!