motioneye-project/motioneyeos #964: Reboot loop after Firewall configuration

Laksuu, 2017-05-24:
Hi. I have installed Motioneyeos on Raspberry Pi 3B model. After basic setup i wanted to make firewall rules with iptables. I have done that in script in **/data/etc/userinit.sh** and it looks like this:

#!/bin/bash

IPTABLES=/usr/sbin/iptables

$IPTABLES -P OUTPUT ACCEPT
$IPTABLES -P INPUT DROP
$IPTABLES -P FORWARD DROP

$IPTABLES -A INPUT -m state --state RELATED,ESTABLISHED -j ACCEPT

$IPTABLES -s publicipaddress -A INPUT -p tcp --dport 9999 -j ACCEPT

$IPTABLES -s 192.168.0.20 -A INPUT -j ACCEPT
$IPTABLES -s 192.168.0.1 -A INPUT -j ACCEPT

Now my firewall rules are staying in iptables after reboot and everything works (blocks and allow wanted traffic throw) but, Motioneyeos is rebooting 3-5min after startup. Its like a loop or something. My guess that it is because some automatic script on Motioneyeos but which one? I have  turned off in Expert Settings  Network Link Watch and Connectivity Watch but it seems that those aren´t the one what causes this problem.

Do you have any idea what could cause this problem?

Here is some Motioneyeos Logs about the problem:

Motioneyeos.log
2017-05-24 15:24:50: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 9999
2017-05-24 15:25:00: [motioneye]  WARNING: Connect error on fd 15: ETIMEDOUT
2017-05-24 15:25:20: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 9999
2017-05-24 15:25:30: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 9999
2017-05-24 15:25:40: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 9999
2017-05-24 15:25:50: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 9999
2017-05-24 15:25:52: [motioneye]     INFO: interrupt signal received, shutting down...
2017-05-24 15:25:52: [motioneye]     INFO: server stopped
2017-05-24 15:25:52: [motioneye]     INFO: tasks stopped
2017-05-24 15:25:54: [motioneye]     INFO: motion stopped
2017-05-24 15:25:54: [motioneye]     INFO: smb mounts stopped
2017-05-24 15:25:54: [motioneye]     INFO: bye!

Messages.log
May 24 15:22:56 192 auth.info sshd[1059]: WARNING: could not open /data/etc/moduli (No such file or directory), using fixed modulus
May 24 15:23:10 192 auth.info sshd[1059]: Accepted password for admin from 192.168.0.20 port 56503 ssh2
May 24 15:25:49 192 user.notice motioneye: not responding for 120 seconds, rebooting
May 24 12:25:49 192 daemon.info : starting pid 1152, tty '': '/etc/init.d/rcK'
May 24 15:25:54 192 auth.info sshd[880]: Received signal 15; terminating.
May 24 15:25:54 192 auth.info sshd[1059]: Exiting on signal 15

ccrisan, 2017-05-24:
You're missing a rule that allows your unit to talk to itself :)

    iptables -A INPUT -i lo -j ACCEPT

Laksuu, 2017-05-24:
You are right. Thanks. Im kind of new to Linux world so making amateur problems are my speciality. 

Donates coming up, after i have completed my project and transported it in real use from test environment!

Thanks again for quick answer! Great job!

ccrisan, 2017-05-24:
You're welcomed. Can we close the issue now?

Laksuu, 2017-05-24:
Yes.

Laksuu, 2017-05-24:
Yes.

closed 2017-05-24 by ccrisan