motioneye-project/motioneyeos #965: Stream freezing, failed to write jpeg data

lexcyn, 2017-05-24:
I'm thinking this could be an SD card issue but wanted to see if anyone else had this happen. Basically the stream freezes constantly or just stops. I checked streameye.log and there are a ton of these:

2017-05-24 11:50:24: INFO : new client connection from 192.168.86.68:37336
2017-05-24 11:50:27: ERROR: 192.168.86.68:37334: not all data could be written
2017-05-24 11:50:27: ERROR: 192.168.86.68:37334: failed to write jpeg data
2017-05-24 11:50:44: ERROR: 192.168.86.68:37336: not all data could be written
2017-05-24 11:50:44: ERROR: 192.168.86.68:37336: failed to write jpeg data

Any ideas?

lexcyn, 2017-05-24:
A little bit more info on my setup:
RPI Zero W with NoIR camera, fast network stream. OC set to PiZero profile. By all accounts the stream should be fine.

bortek, 2017-06-01:
check your sd card to start with, check your power support if it provides enough power. Google for problem like yours. This has nothing to do with motioneyeos

closed 2017-06-01 by bortek