motioneye-project/motioneyeos #967: motion for raspberry pi not compiled with WITH_MMAL flag

FaBjE, 2017-05-25:
This is more of a request than a bug but I found the following.
My raspberry pi camera is going kind of bazerk with the exposure and auto white balance. 
To fix this I need to disable it. by passing the:
_mmalcam_control_params -ex night
mmalcam_control_params -awb sun_
Parmameters to motion, however motion logs that it does not know these parameters. I check the motion version and it is the latest one.

motionEye Version 	0.35.2
Motion Version 	4.0.1+git37b3595
OS Version 	motionEyeOS 20170329

I did some digging in the makefile of the motion project and found that you need to specifically enable the MMAL support in the build process. (At least that is what I understand from https://github.com/Motion-Project/motion/blob/master/CMakeLists.txt)
Therefore I suspect it is not enabled in the motion build for the latest Raspberry Pi motioneye OS 

**TL; DR;**
Could you please enable the WITH_MMAL flag in motion for the RPI?

jogu, 2017-05-26:
If mmal support wasn't enabled you wouldn't get any image at all from the raspberry pi camera.

FaBjE, 2017-05-26:
Ah, I thought it only was a way to set camera settings. In that case it must be a motion thing.
Case closed for here :)

closed 2017-05-26 by FaBjE