motioneye-project/motioneyeos #969: Slow video, low fps... horrible delay.

topdiogo, 2017-05-26:
Hello, i have a brand new Rapberry Pi3 with a ip camera, setup with motioneyeos. I think my configs are ok, as shown on the link, but the video feed is horrible, more than 10 sec delay... if i set up higher resolution is worst, not to mention the maximum it accepts is 1020*768...
I think there is something wrong with the fps. Could you please help?
Configs print: [http://i.imgur.com/ORyYVio.jpg[/img](url)
P.S. Everything works very smooth using the CMS ou Ispy software.

bortek, 2017-06-01:
It works fine for me and many others and delay is part of the system. High resolution will always case delays and slow performance. sorry cant help more. Try googling or use some other software.

closed 2017-06-01 by bortek

topdiogo, 2017-06-01:
this sucks. No one helps! Shitty community!!