motioneye-project/motioneyeos #973: RPI zero - Fast Network Camera - Best Settings

Gman1988, 2017-05-30:
Hi, i anyone knew what is the optimal settings to get optimal performance in FNC mode?

I already have 256 ram for GPU and overclocked using settings PI ZERO.

Any other improvements?
More ram for GPU will change anything?

ccrisan, 2017-05-30:
Reduce the image quality to something between 10 and 20 and I think you'll be pleased with the outcome.

Gman1988, 2017-05-30:
Wow, thanks for quick answer, but i need to ask stupid question ;]

What is the difference between resolution and image quality ? I mean the impact on the image i can see ?

The amount of the RAM for GPU have also impact ?

And does in the FNC mode "monitoring system" settings working ? because i dont see any effect on the image.

ccrisan, 2017-05-31:
Your questions are not stupid. Resolution refers to the number of pixels that the image is composed of (its actual dimensions), while the quality refers to the overall JPEG quality. Here's an image I found googling that should somehow exemplify this quality: http://help.adobe.com/en_US/fireworks/cs/using/images/oe_jpg_loss.png. From my experience with FNC mode, a quality of 10 to 20 is still acceptable.

The amount of GPU memory (and the GPU itself) doesn't have any impact on the performance so don't bother tweaking it.

Many features, including the monitoring info, are not available in FNC mode, in order to improve the performance.

closed 2017-05-31 by ccrisan