motioneye-project/motioneyeos #974: Pi Zero W + Camera Modul V2

Noxigin, 2017-05-31:
Hello,

i got a Raspberry Pi Zero W with the camera module v2. I installed motioneyeos (20170329) on it and everything works fine. Love it.
Now i tried to switch to the "Fast Network Camera" and i got no preview anymore. There is only a grey screen with a camera symbol.
I read a lot of threads about similar problems and i tried a new Installation, a different power supply, with and without wifi and i changed the ports. Nothing helps.

If i deactivate Fast Network Camera everything works fine again.

Can someone help me? I don't got the skills to get some logs, so if it's necessary please tell me how.

Thanks a lot

bortek, 2017-06-01:
Please post your question on the forum page instead of opening an issue.

closed 2017-06-01 by bortek

ccrisan, 2017-06-01:
@Noxigin this may be due to the fact that your browser doesn't render MJPEGs or that connections on port 8081 cannot be made.

Noxigin, 2017-06-01:
I tried with IE11. I will try with another browser. The port i have changed, so that i think that this is not the reason.
I also found a thread in the forum with Problems with static ip and no video. I also have static ip configuration. I will check this, too.

Should i open a thread in the forum or is it here okay?

Thanks bortek for the tip with the forum and thank ccrisan for your response.

ccrisan, 2017-06-01:
The forum is always a good idea. It has a wider visibility.

Noxigin, 2017-06-01:
ok, if the problem persits i will post it there. 

Thank for your fast support.

Noxigin, 2017-06-01:
Firefox works! :) Sometimes it can be so easy. Thank you!