motioneye-project/motioneyeos #975: My security monitor keeps going offline

gigawert, 2017-05-31:
At first my monitor works fine. But after a couple weeks of my RPi 2 with motioneyeOS + camera running, it doesn't work anymore. And every time I check to see why, it is because it expected ethernet. But I already set it to use the wireless network multiple times before. Anyone know why this is?

bortek, 2017-06-01:
Please test the latest on pi zero if you have one.

gigawert, 2017-06-01:
I do not have a pi zero, sorry

falcondan, 2017-06-04:
I tried updating my pi2 to the motionEyeOS 20170329 through the updater within Motioneye. It,broke my configuration. Plugging in a monitor showed that it wouldn't even boot up again. I was able to re-image my sd card with the instructions here: "https://github.com/ccrisan/motioneyeos/wiki/Installation" and then load a backup configuration i had from a few months ago (even though it said it wasn't successful, it seems that it was). Can you ping the address that the board has?

gigawert, 2017-06-05:
I can't find the address in my router page because it won't even show as online in the network.

falcondan, 2017-06-06:
Can you plug a monitor and keyboard in to see what's happening? Does the board manage to boot at all when it gets power? Does it ever take you to a command line? If it does, run "ifconfig" that will show what kind of address the board has, if any.

gigawert, 2017-06-06:
The screen is blank when I plug it in, even after rebooting the Pi.

falcondan, 2017-06-06:
Would it be possible to try loading another OS on the Pi? Just to verify that the board is still functional? Maybe https://www.raspberrypi.org/downloads/raspbian/  If that doesn't load, then I think it has to be either the SD cards or the Pi. The 20170329 version of MotionEyeOS definitely worked on my Pi2...After I re-imaged my card.

gigawert, 2017-06-07:
I wrote the image to the microSD card again and it works now. Thanks anyway for your time! :+1:

falcondan, 2017-06-08:
Glad it's working! I had to check to see which version of Win32diskimager I have installed on my Windows 10 PC. It looks like version 0.9.5. Definitely something to keep in mind, I didn't even think about that being a possible culprit.

jasaw, 2017-08-07:
Looks like a Win32diskimager problem. Closing issue.

closed 2017-08-07 by jasaw