motioneye-project/motioneyeos #976: MotionEye crashes only with camera plugged in

Acaravet, 2017-06-01:
I can start and run motioneye just fine without my camera plugged in. I can change the network settings  and everything but once I plug in my usb logitech c920 a few seconds later it crashes and starts the boot loop.
Also when I try to ssh into it it asks for a password but no user name. I have tried the password I set in the browser and a bunch of default pi passwords and nothing works. I have tried through a mac terminal and through windows running putty with the same results. 
How do you ssh in?

Any help would be greatly appreciated. (very frustrating!!)

Thanks

bortek, 2017-06-01:
power or some other non motioneyeos related issue. Google is your friend. You could try posting your question to forum page too.

closed 2017-06-01 by bortek

Acaravet, 2017-06-04:
I can't find anything that says what the password is for SSH into motion eye other than people sing to use the same password as I set in the UI. which doesn't work for me.