motioneye-project/motioneyeos #982: I just changed the os_networkless variable and now can't connect

imirah, 2017-06-04:
I wanted to have motioned record to SD card if the router goes down...so I did as you mentioned i.e. change the variable to true.
However, now I can't connect to motioneye. Is there a way I can change the variable back and what is the solution to the issue? thanks

ccrisan, 2017-06-04:
That setting will disable everything related to networking. You can't have motionEyeOS work both in network and networkless mode, depending on whether you currently have or not a working connection. Sorry about that.

closed 2017-06-04 by ccrisan

imirah, 2017-06-04:
How do I change it back. I can't ssh to the Pi now

Imran

ccrisan, 2017-06-04:
Since your unit no longer has any network connection, you'll need to take your SD card out and put it into your PC.

imirah, 2017-06-04:
ok , Ive inserted into my mac...not sure how I get to the /data/etc/os.conf file.
All i can see is 
bcm2710-rpi-3-b.dtb
bootcode.bin
cmdlin.txt
config.txt
fixup.dat
fwupdater.gz
kernel.img
start.elf