motioneye-project/motioneyeos #983: Option to disable HDMI output

MatthewSteeples, 2017-06-05:
Supposedly you can save 25mA of power by disabling the HDMI output

`/usr/bin/tvservice -o`

This needs to be done on each boot, so wants adding to init.d or equivalent. This is more of a placeholder issue until I work out how to put together some instructions to configure this over SSH

Chiny91, 2017-06-06:
Hmm yes, something happens anyway:
```
#tvservice -s
state 0x40001 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
#tvservice -o
Powering off HDMI
#tvservice -s
state 0x120001 [TV is off]
```
I've nothing attached to the HDMI output, so cannot verify whether this actually works nor whether HDMI can be re-enabled.

Where does the 25 mA come from ?

Doing this every boot seems easy enough according to:
https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands

bortek, 2017-06-12:
@MatthewSteeples I wouldn't even consider saving that little amount, actually nothing below 100mA. But if you want it please consider the suggestion of Chiny91 and then close this issue. thanks.

MatthewSteeples, 2017-06-19:
@Chiny91 Thanks for that. Ashamed I hadn't read the docs!

@bortek Normally I'd agree, but when solar powering a Pi, every mA helps! :)

closed 2017-06-19 by MatthewSteeples