motioneye-project/motioneyeos #984: Macroblocking in recorded video - Stream looks fine

anderson620, 2017-06-05:
Setup: 
Raspberry Pi Zero W - latest MotionEyeOS version (20170329) - FPC camera set to:
FNC
1024x768
20 FPS (Have tried every combination from 5-30 FPS)
20% Image quality (Have tried every setting from 5-100%)
No overclocking
Static IP over WiFi

Full desktop (SSD, 16GB RAM) running Ubuntu 16.04 with MotionEye installed per instructions here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu
Not currently used for anything other than MotionEye (no games, browsing, etc.)

No Motion detection, FTP, SSH, Web Server, etc. turned on
Continuous Recording
Static IP over Ethernet
Movie Quality 100%

Issue:
The stream is perfect but the resultant videos are overly pixelated and choppy (played in VLC)
I have tried different settings for everything listed above in every combination possible. The stream definitely shows variations depending on the settings but the recorded footage does not change. I have tried to take as much load off of the RPi Zero as possible by having the desktop handle any rotation, etc.

Is there any way to store output that is of the quality as the stream? I intend to use the setup as home security but if I ever needed to play back video persons would not be identifiable. 
Any help would be greatly appreciated.

fjibril, 2017-06-12:
Hello, I thnik this might be due to the hardware used.
I also found mostly blocky image captures when I tried to setup motion detection on the cheap using a RasbPi Zero W. But in addition I also got lag of several seconds in what I could see in the UI and the movement (was viewing myself). Captured images on motion detection where mostly garbled. In order to remedy I bought a Raspberry Pi 3 with more oomph, and not only did I get rid of the lag, but also of the artefacts in captures. So I guess that might have to do with it. Also, I think 20FPS is very high. The lag I saw was even when trying to make it work with only 2 FPS.

anderson620, 2017-06-12:
Thank you for the reply. This is something I've seen in the past, although I don't think this is the case now as I am not using motion detection at all. The RPi Zero itself seems fine for the settings listed above as even if the stream is being watched it looks fine (which in itself can be taxing on the system). It wasn't mentioned in the original post but the same issue occurs when I used my RPi 3 on place of the Zero. It's not the hardware but something on the software level as it only affects the recorded video.

bortek, 2017-06-12:
Are you running MotionEye or MotionEyeOS? These are kinda separate things. Are you using Pi Camera or some other , If not Pi camera then there can be anything else so you will have to troubleshoot it. ALso 20 FPS is waaaay to high, try lowering it to 5.

anderson620, 2017-06-12:
Thank you for the response. I do appreciate all of the input, although I have found the issue and this case can be closed. 
The issue isn't with the settings listed above ( you'll note that I mentioned using MotionEye on Ubuntu machine and MotionEyeOS on Zero.) In fact, with Zero set as Fast Network Camera 30 fps is very possible. 
The issue was with VLC and it's Hardware-accelerated video encoding. Switching to use SMPlayer has corrected the issue. 
I would imagine (although I could be wrong) that if the issue was with the underpowered nature of the Zero then issues would have been present in the stream and not just the recordings. Of course 20 fps would be way too high for any usable feed if motion detection were enabled, but that is very expensive overhead that I do not find useful given the degradation in quality. 
Again, thank you both for the input.

closed 2017-06-12 by anderson620