motioneye-project/motioneyeos #985: Reapberry Pi3 WiFi wpa_supplicant: failed (brcmfmac)

mwsl, 2017-06-06:
I have a Raspberry Pi 3 running Raspbian that uses the built-in WiFi adapter to connect to my router.  No connection problems.  However whenI try and connect using the same interface on MotionEyeOS, I get a "wpa_supplicant: failed (brcmfmac)" when the system is booting.  The Ethernet connection works correctly.  My boot.log says this:

```
---- booting motionEyeOS 20170329----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting watchdog: done
 * Starting eudev: done
 * Setting root password: done
 * Starting wpa_supplicant: failed (brcmfmac)
 * Configuring wireless network: no link
 * Configuring wired network: dhcp
 * Setting current date using http: Tue Jun 6 07:01:52 EDT 2017
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.105/24
 # Default gateway is 192.168.1.1
 # DNS server address is 192.168.1.1
```

My /data/etc/wpa_supplicant.conf file is configured as follows:

```
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
    scan_ssid=1
    psk="PASSWORD"
    ssid="MWSLN"
}
```

I've tried all of the troubleshooting suggestions here: https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

What exactly does **brcmfmac** mean?

Do you have any suggestions on how to correct this?

Thanks.

Chiny91, 2017-06-06:
Judging by past similar issues, you can fidget about for a while and be lucky, or not, in getting a fix.   It might be quicker to re-image the SDcard and try again.   I'd give the SDcard a good thump by (1) re-formatting with any formatter to hand, and (2) testing the card with an SDcard test program (I use F3 on a Mac but others exist for other OSs).

brcmfmac is the driver for the Broadcom chip.   I guess it stands for BRoadCoM _mumble_ Media Access Control.

mwsl, 2017-06-06:
Thanks.

I used diskpart in Windows to do a CLEAN and FORMAT and then re-imaged the card.  I now have an IP address on a router that temporarily has no password.  Now I can change things one step at a time until I get back to a secure system with a WiFi connection.

When it says that the SSID and PASSWORD should have no "special characters", what is considered a special character?  Is a minus sign "-" or a parentheses "(" a special character, or do you mean non-printing characters like TAB.

Must the SSID and PASSWORD only contain letters and numerals?

Chiny91, 2017-06-07:
ssid/password; presumably you mean router parameters.

As the ssid is publicly visible, convention says that use of special characters is pointless and it is best to use alphanumerics.   Most routers are Linux based without PAM (pluggable authentication modules), so convention dictates a 6 (or longer) alphanumeric password.   Of course, for some setups, possibly not old Windows machines or ultra-cheap routers, special characters will work.   However, if it is not just a single machine owned by you, caution makes for an easy life.

I suspect motionEyeOS can be persuaded to send special characters for ssid/password but I'm not going to test it to avoid being locked out.   That need not deter your experimentation 😄

mwsl, 2017-06-07:
I did get everything working correctly.  I was able to put the hyphen back into my **ssid** and add some punctuation characters to the **psk** so it matched what the other Raspberry Pi computers on the net were already configured for,  However, when switching back from no password to a password (using the MotionEyeOS web) interface it initially wouldn't work with the router set to wpa/wpa2 encryption.  I checked the **wpa_supplicant.conf** file and saw that it had the **ssid** and **psk** values set correctly but if I remember correctly there was also a line that said **key_mgmt=NONE**.  I deleted that line, rebooted and it worked.  I guess if the initial setup is for no password/encryption, and then you add a password, MotionEyeOS doesn't know that you turned on security on the router.

Thanks for fixing my problem.

Chiny91, 2017-06-07:
Glad that is fixed.   Just to tidy/close this query, let us consider what happened.

**key_mgmt=NONE** in **wpa_supplicant.conf** 
According to [http://www.cs.upc.edu/lclsi/Manuales/wireless/files/wpa_supplicant.conf](url);
**NONE = Use only Group Keys (deprecated**...
which is not wanted as [motionEyeOS enforces WPA/WPA2](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems).
Deleting that line will (first URL):
**If not set, this defaults to: WPA-PSK WPA-EAP**
which is what is required by motionEyeOS.

mwsl, 2017-06-08:
What I was trying to say is that somehow in the setup process **key_mgmt=NONE** was added to the  **wpa_supplicant.conf** file by MotionEyeOS.  It didn't get there by me editing the file.  Perhaps there should be a check in the code so that if a **psk** value is specified, then **key_mgmt=NONE** gets deleted.

closed 2017-06-08 by mwsl

OFark, 2018-07-11:
Problem is when you connect to a password free wireless connection it adds an option to the /data/etc/wpa_supplicant.conf
"key_mgmt=NONE"
When you set a password again, it doesn't clear this option, and it then never passes a psk with authentication. Remove that option and you'll be fine.