motioneye-project/motioneyeos #986: Central server motion and capture

timdonovanuk, 2017-06-07:
I guess I'm majorly not understanding something, but in the [central server use case](https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios#multiple-devices-with-a-central-server)  you lose all motion and capture capability. Why can the central server not just run motion on the stream it receives from the network cameras, and record the frames appropriately? 

Thanks.

ccrisan, 2017-06-07:
That's exactly what the central server does.

> The server can be motionEyeOS-based (preferably running on a multicore board) or it can be a regular Linux system that runs motionEye.

You don't lose motion/capture capabilities, you just move them from individual cameras to a central server and allow individual cameras to run other stuff than motionEye.

closed 2017-06-07 by ccrisan

timdonovanuk, 2017-06-07:
Ah my apologies! I think when I tried the motion option was not visible for some reason. I must have messed something up. Maybe add "offload motion processing" as an advantage to the Usage-Scenarios.

Cheers :)