motioneye-project/motioneyeos #987: Show Frame Changes in % not pixel amount

timdonovanuk, 2017-06-07:
At the moment with "Show Frame Changes" on it shows the number of changed pixels. This would be more valuable to users if it showed the %, as it would match up with the "Frame Change Threshold" slider, and allow you to tweak this value appropriately. 

Presumably its just a case of doing = (changed pixels/camera resolution)*100

ccrisan, 2017-06-08:
This can't be done in motionEye. It's motion-specific.

closed 2017-06-08 by ccrisan

timdonovanuk, 2017-06-08:
Perhaps another solution then:

- change the Frame Change Threshold slider to have a tooltip (or label) to show the % expressed as pixels
- show Frame Change Threshold expressed as pixels as a text overlay on the video (when enabled). This would be better as it would allow you to compare the number in the top right with say, this new overlay in the top left.

This would be so incredibly useful to tweak the threshold. At the moment you have to randomly set it, comparing apples to oranges.

Both my ideas could be triggered only when the "Show Frame Changes" is enabled. 

Thanks.

timdonovanuk, 2017-06-08:
I have got this working using the list of modifiers here http://www.lavrsen.dk/foswiki/bin/view/Motion/ConversionSpecifiers. %o shows the threshold and %D the number of pixels detected as Motion

e.g. by setting the Text Overlay field to %D / %o

As an idea, when the user selects Show Frame Changes instead of doing text_changes = on, just append %D / %o to the start of their left text label. For example, the label would be "Camera 1" and then changed to "%D / %o\nCamera 1". This results in:

[image]

changing to:

[image]

which I think is a very nice solution.