motioneye-project/motioneyeos #988: Problem with VR CAM

Lakemast, 2017-06-07:
Hello,
I want to add a camera that gives me only plain jpeg images, but the motioneyeOs says that it is
a not supported camera

[image]

The URL for getting plain jpeg images is: "_http://192.168.1.112/snapshot.jpg?user=admin&pwd=admin&strm=1_"

This is the camera i want to add  [VR CAM P2](http://www.juancctv.com/productShow.asp?ArtID=660)

When I put this jpeg url  "_http://192.168.1.112/snapshot.jpg?user=admin&pwd=admin&strm=1_" in my web browser, I can download the jpeg image.

I also tried to add the camera to ispy software to see if the camera was really working and it worked

[image]

What am I doing wrong ?

Thanks in advance!

ccrisan, 2017-06-08:
If it says that's not a supported network camera, then the URL you're using does not render a simple JPEG but most likely some HTML page that wraps your JPEG. Make sure to use a URL that renders the final JPEG. It helps opening the URL with your browser and playing with the DOM inspection tools to identify the URL.

closed 2017-06-08 by ccrisan

Lakemast, 2017-06-08:
Sorry, but i guess i didn't explain in a right way.

When I put this in my web browser: "_http://192.168.1.112/snapshot.jpg?user=admin&pwd=admin&strm=1_"

I got only the picture, with no html view, like this:

[image]

ccrisan, 2017-06-08:
Hmm so the camera does not directly serve the image but offers it for download. I'm not sure if that works with motion.

Lakemast, 2017-06-08:
Thank you ccrisan for your quick answer!
Congratulations for making this great application.
I will try another camera instead.
Thank you!