motioneye-project/motioneyeos #992: Limit the number of days when viewing photos (Question)

Jasonb11, 2017-06-09:
Is there a way to limit the number of days when viewing photos? I am getting time outs when trying to view previous days because Motioneyeos is trying to look through over a month of photos taken in 1 minute intervals.

bortek, 2017-06-12:
I have the same experience and I only set rotation for 1 month. So I just wait patiently for a minute and then it shows up. So my suggestion to you would be to get a faster sdcard for now.

fredericcollin, 2017-06-23:
Same problem for me. Bortek I don't think it is related to SD card speed because all my picture and video data is saved on a separate hard drive. When there are lets say more than 100 videos per day or a few thousand pictures per day, Motioneye times out when I want to access these. Then the only way to get to these files is per FTP/SSH or disconnect the drive and access it directly.