motioneye-project/motioneyeos #993: Using Google Drive

Dobi29, 2017-06-10:
Hi,
i'm using the upload to Google Drive for one year. Since two days it doens'nt work. When testing the service i takes more than an minute, getting the message "error occurred. refreshing the system"Thx
Thx
Klaus

Chiny91, 2017-06-10:
Have you tried refreshing the page with your browser ?   A proper _flush the cache_ refresh may be required.   To be convincing, another browser or another computer/device would be useful.

Dobi29, 2017-06-10:
yes, and i tried it with three devices in two networks....

ccrisan, 2017-06-10:
@Dobi29 please attach your motioneye.log file here.

Dobi29, 2017-06-10:
here is my log file

[motioneyelog.txt](https://github.com/ccrisan/motioneyeos/files/1066107/motioneyelog.txt)

ccrisan, 2017-06-11:
For some unknown reason, the file `/data/etc/uploadservices.json` becomes empty and thus loses the upload service state (token, location and whatnot) that is required for the upload service to work. You're not the first one to report this problem but each time I hear about this, I have a hard time understanding how the heck that state file became empty (or corrupted).

Is this the entire log file? Could you maybe log in remotely and try to obtain the rotated files (`/var/log/motioneye.log.1.gz` and similar)?

Anyways, the solution for now is to reenable the upload service and configure it from scratch.

closed 2017-06-12 by bortek