motioneye-project/motioneyeos #994: Smb share won't connect

silverbullettruck2001, 2017-06-11:
My Pi3 stopped recording and writing images to the remote file share I have setup.  I haven't changed anything on the file server.  Here is the error I am seeing in the logs after a fresh reboot.  It says the device or resource is busy, but I am not sure why.

2017-06-11 08:13:36: [motioneye]     INFO: rebooting
2017-06-11 08:13:36: [motioneye]     INFO: executing "/sbin/reboot"
2017-06-11 08:13:37: [motioneye]     INFO: interrupt signal received, shutting down...
2017-06-11 08:13:37: [motioneye]     INFO: server stopped
2017-06-11 08:13:37: [motioneye]     INFO: tasks stopped
2017-06-11 08:13:39: [motioneye]     INFO: motion stopped
2017-06-11 08:13:39: [motioneye]     INFO: smb mounts stopped
2017-06-11 08:13:39: [motioneye]     INFO: bye!
2017-06-11 08:13:53: [motioneye]     INFO: hello! this is motionEye server 0.35.2
2017-06-11 08:14:01: [motioneye]     INFO: cleanup started
2017-06-11 08:14:01: [motioneye]     INFO: wsswitch started
2017-06-11 08:14:01: [motioneye]     INFO: tasks started
2017-06-11 08:14:01: [motioneye]     INFO: mjpg client garbage collector started
2017-06-11 08:14:01: [motioneye]     INFO: smb mounts started
2017-06-11 08:14:01: [motioneye]     INFO: server started
2017-06-11 08:14:05: [motioneye]  WARNING: 403 GET /login/?_=1497186843771&_username=_&_signature=7a613f60b80f9137e8e0cf853f5422d2153dc8a5 (192.168.0.70) 2.35ms
2017-06-11 08:19:07: [motioneye]    ERROR: failed to remove smb mount point "/data/media/motioneye_192_168_0_24_freenas_cam": [Errno 16] Device or resource busy: '/data/media/motioneye_192_168_0_24_freenas_cam'
2017-06-11 08:23:24: [motioneye]    ERROR: failed to remove smb mount point "/data/media/motioneye_192_168_0_24_freenas_cam": [Errno 16] Device or resource busy: '/data/media/motioneye_192_168_0_24_freenas_cam'
2017-06-11 08:23:55: [motioneye]    ERROR: failed to unmount smb share "//192.168.0.24/freenas/Cam" from "/data/media/motioneye_192_168_0_24_freenas_cam"

bortek, 2017-06-12:
You have WARNING: 403 GET /login/? and then you have  Device or resource busy: 

Do you mind checking your NFS server, reboot it, make sure you can access it from other computer, then reboot motioneyeOS completely (reboot Pi) and see what happens. 

Kinda basic troubleshooting that you should be able to do instead of opening  an Issue.

silverbullettruck2001, 2017-06-12:
I have already rebooted the pi, as indicated by the logs.  I am not sure why I would need to reboot the entire file server when the serveracity is operating fine.  I would prefer to not reboot it unless there is a specific reason why as it will affect othe users that access it.

I have tried unmounting the file system from the motioneye as well and it won't allow me to do that either.  I have also completely powered off the Pi waited 10 seconds and plugged it back in.  Nothing seems to allow it to mount or unmount the file server file system.  Thoughts?

Thanks for your help!

silverbullettruck2001, 2017-06-16:
Any other suggestions on how to fix this?

Xris2017, 2017-07-20:
I have a similar error with motionEye 0.36.1 (motionEyeOS 20170710) on a raspi3 completely new installed on a new SDCard yesterday connected with a network storage over wifi.

If think there must be a bug because if I start the raspi all works fine for about ~5 minutes.
But everytime after ~5 minutes the video on webgui stops with several messages in motioneye.log like:

[motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED

It seems there is a job that tries to unmount the network storage in an interval without luck because it is in use. Then it tries to mount without luck because it is already mounted.

Tried to read and write to the mounted dir without problems.

Maybe this log can help to solve this problem:

login as: admin
Welcome to Cam!
admin@192.168.0.16's password:
[root@Cam ~]# tail -f /var/log/motioneye.log
2017-07-20 23:15:44: [motioneye]     INFO: rebooting
2017-07-20 23:15:44: [motioneye]     INFO: executing "/sbin/reboot"
2017-07-20 23:16:48: [motioneye]     INFO: hello! this is motionEye server 0.36.1
2017-07-20 23:16:52: [motioneye]     INFO: cleanup started
2017-07-20 23:16:52: [motioneye]     INFO: wsswitch started
2017-07-20 23:16:52: [motioneye]     INFO: tasks started
2017-07-20 23:16:52: [motioneye]     INFO: mjpg client garbage collector started
2017-07-20 23:16:52: [motioneye]     INFO: smb mounts started
2017-07-20 23:16:52: [motioneye]     INFO: server started
2017-07-20 23:17:13: [motioneye]  WARNING: 403 GET /login/?_=1500585431951 (192.168.0.4) 4.58ms
2017-07-20 23:21:52: [motioneye]    ERROR: failed to unmount smb share "//192.168.0.3/cam" from "/data/media/motioneye_192_168_0_3_cam"
2017-07-20 23:26:56: [motioneye]    ERROR: failed to mount smb share "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam"
2017-07-20 23:26:56: [motioneye]    ERROR: failed to unmount smb share "//192.168.0.3/cam" from "/data/media/motioneye_192_168_0_3_cam"
2017-07-20 23:27:00: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:00: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:00: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:00: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:01: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:01: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:01: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:01: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:01: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:02: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-20 23:27:02: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
^C
[root@Cam ~]# cd /data/media/motioneye_192_168_0_3_cam_cam
[root@Cam motioneye_192_168_0_3_cam_cam]# ls -la
total 4
drwxrwxrwx    3 root     root             0 Jul 20 23:21 ./
drwxrwxr-x    4 root     root          4096 Jul 20 19:32 ../
-rwxrwxrwx    1 1040     users            0 Jul 20 23:17 .keep*
drwxrwxrwx    2 1040     users            0 Jul 20 19:16 2017-07-20/
[root@Cam motioneye_192_168_0_3_cam_cam]# touch test
[root@Cam motioneye_192_168_0_3_cam_cam]# ls -la
total 4
drwxrwxrwx    3 root     root             0 Jul 20 23:27 ./
drwxrwxr-x    4 root     root          4096 Jul 20 19:32 ../
-rwxrwxrwx    1 1040     users            0 Jul 20 23:17 .keep*
drwxrwxrwx    2 1040     users            0 Jul 20 19:16 2017-07-20/
-rwxrwxrwx    1 1040     users            0 Jul 20 23:27 test*
[root@Cam motioneye_192_168_0_3_cam_cam]# rm test
[root@Cam motioneye_192_168_0_3_cam_cam]# ls -la
total 4
drwxrwxrwx    3 root     root             0 Jul 20 23:27 ./
drwxrwxr-x    4 root     root          4096 Jul 20 19:32 ../
-rwxrwxrwx    1 1040     users            0 Jul 20 23:17 .keep*
drwxrwxrwx    2 1040     users            0 Jul 20 19:16 2017-07-20/
[root@Cam motioneye_192_168_0_3_cam_cam]#

Beste regards

ccrisan, 2017-07-21:
@Xris2017 please enable debugging and try reproducing the problem. The log file will contain details as to why it wants to unmount the smb share.

Xris2017, 2017-07-22:
@ccrisan started motioneyeos with debugging mode enabled. Same problem after ~5 minutes.
The problem starts when motioneyeos checks the smb mountpoints.

I also checked the mountpoint after the failure starts and it is ready and up (read and writeable)....

...
2017-07-22 15:18:15: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 3.03ms
2017-07-22 15:18:21: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 2.81ms
2017-07-22 15:18:22: [motioneye]    DEBUG: motion detection is disabled for camera with id 1
2017-07-22 15:18:26: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 2.93ms
2017-07-22 15:18:31: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 2.90ms
2017-07-22 15:18:32: [motioneye]    DEBUG: motion detection is disabled for camera with id 1
**2017-07-22 15:18:34: [motioneye]    DEBUG: checking SMB mounts...
2017-07-22 15:18:34: [motioneye]    DEBUG: listing smb mounts...
2017-07-22 15:18:34: [motioneye]    DEBUG: found smb mount "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam"
2017-07-22 15:18:34: [motioneye]    DEBUG: found smb mount "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam"
2017-07-22 15:18:34: [motioneye]    DEBUG: making sure mount point "/data/media/motioneye_192_168_0_3_cam_cam" exists
2017-07-22 15:18:34: [motioneye]    DEBUG: mounting "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam" (sec=None)
2017-07-22 15:18:34: [motioneye]    DEBUG: mounting "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam" (sec=ntlm)
2017-07-22 15:18:34: [motioneye]    DEBUG: mounting "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam" (sec=ntlmv2)
2017-07-22 15:18:34: [motioneye]    DEBUG: mounting "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam" (sec=ntlmv2i)
2017-07-22 15:18:34: [motioneye]    DEBUG: mounting "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam" (sec=ntlmsspi)
2017-07-22 15:18:34: [motioneye]    DEBUG: mounting "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam" (sec=none)
2017-07-22 15:18:34: [motioneye]    ERROR: failed to mount smb share "//192.168.0.3/cam" at "/data/media/motioneye_192_168_0_3_cam_cam"
2017-07-22 15:18:34: [motioneye]    DEBUG: unmounting "//192.168.0.3/cam" from "/data/media/motioneye_192_168_0_3_cam"
2017-07-22 15:18:34: [motioneye]    ERROR: failed to unmount smb share "//192.168.0.3/cam" from "/data/media/motioneye_192_168_0_3_cam"
2017-07-22 15:18:34: [motioneye]    DEBUG: stopping motion**
2017-07-22 15:18:36: [motioneye]    DEBUG: recevied relayed event stop for thread id 1 (camera id 1)
2017-07-22 15:18:36: [motioneye]    DEBUG: clearing motion detected for camera with id 1
2017-07-22 15:18:36: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=69c3eff4dbcf307b62d587df184d82bda0a3cbd1 (127.0.0.1) 7.87ms
2017-07-22 15:18:36: [motioneye]    DEBUG: recevied relayed event stop for thread id 1 (camera id 1)
2017-07-22 15:18:36: [motioneye]    DEBUG: clearing motion detected for camera with id 1
2017-07-22 15:18:36: [motioneye]    DEBUG: 200 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=69c3eff4dbcf307b62d587df184d82bda0a3cbd1 (127.0.0.1) 10.25ms
2017-07-22 15:18:36: [motioneye]    DEBUG: 200 HEAD / (127.0.0.1) 6.22ms
2017-07-22 15:18:36: [motioneye]    DEBUG: connection closed for mjpg client for camera 1 on port 8081
2017-07-22 15:18:36: [motioneye]    DEBUG: mjpg client for camera 1 on port 8081 removed
2017-07-22 15:18:36: [motioneye]    DEBUG: creating mjpg client for camera 1
2017-07-22 15:18:36: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2017-07-22 15:18:36: [motioneye]    DEBUG: connection closed for mjpg client for camera 1 on port 8081
2017-07-22 15:18:36: [motioneye]    DEBUG: mjpg client for camera 1 on port 8081 removed
2017-07-22 15:18:37: [motioneye]    DEBUG: creating mjpg client for camera 1
2017-07-22 15:18:37: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-07-22 15:18:37: [motioneye]    DEBUG: connection closed for mjpg client for camera 1 on port 8081
2017-07-22 15:18:37: [motioneye]    DEBUG: mjpg client for camera 1 on port 8081 removed
2017-07-22 15:18:37: [motioneye]    DEBUG: creating mjpg client for camera 1
2017-07-22 15:18:37: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
...

ccrisan, 2017-07-22:
@Xris2017 can I see your thread-1.conf file? Please replace any passwords with asterisks or so. This time please attach it to the commend instead of pasting its content here.

Xris2017, 2017-07-22:
@ccrisan see thread-1.conf attached.

Thanks!
[thread-1.conf.txt](https://github.com/ccrisan/motioneyeos/files/1167531/thread-1.conf.txt)

ccrisan, 2017-07-22:
Could you please try configuring and using a lowercase username? It might be a bug due to using a capital letter in your network share username ("Cam").

Xris2017, 2017-07-22:
Oh my gosh! Works like a charm now...
Thank you so much! You guy rock! 👍

And thanks for this nice piece of software! :)

ccrisan, 2017-08-05:
https://github.com/ccrisan/motioneye/commit/c5875f4544288baa7394ce33126ffa11312af723 fixes this.

closed 2017-08-05 by ccrisan