motioneye-project/motioneyeos #995: Challenges w/ 64GB SD on Raspberry Pi 2

3c509, 2017-06-11:
Using versions 20170329 and 20170611 on Raspberry Pi 2 with SanDisk Ultra 64GB SD card yields same error message at first boot.  "can't open /dev/ttylogin : no such file or directory"

I used the writeimage.sh script to build the SD card per [instructions on the wiki](https://github.com/ccrisan/motioneyeos/wiki/Installation).  I tested with a smaller 32GB SD card and had no problems with version 20170611.

Attached is a photograph of my monitor capturing some of the error messages on the console.

[image]

ccrisan, 2017-06-12:
Most likely there's a problem accessing the SD card, as the first error says "mounting /dev/mmcblk0p3 on /data failed: Invalid argument". That's the root cause of all evil that follows.

At this point I'd recommend trying to install raspbian and see if it can expand the partition to the entire disk space. If yes, get back to us and I'll make sure to update the firmware and the kernel to the latest available with Raspbian.

3c509, 2017-06-12:
Okay will do.  Thanks for the pointers.  I'll let you know how it goes.

jasaw, 2017-08-12:
Any update on how you went with raspbian?

jasaw, 2019-10-30:
Closing issue due to lack of follow up.

closed 2019-10-30 by jasaw