motioneye-project/motioneyeos #996: Unable to open video devis

Sponkenberg, 2017-06-12:
Hi,

Maybe this will be posted on the wrong place... but I can't see the video stream from my ip camera.

It is possible to log on to it via rtsp://192.168.x.x:1024 and user-id and password.

I need help in trouble shooting!

Best regards,
Sponkenberg

Frankthedoc, 2017-06-20:
I had some similar trouble with my Anran H264. My suggestion is to search through your camera manual and find the exact URL for the rtsp stream. Mine looked like this:
rtsp://192.168.x.x:554/h264?user=admin&password=123456&stream=0 
I put all of this URL into motioneyeos "as is" and it worked without needing a separate user/password. 
The camera has two streams, one high resolution and one low resolution. If you don't specify the stream, you won't see anything.

jasaw, 2017-08-17:
Should this information be added to the wiki? Maybe under "How to add remote IP camera"?

ccrisan, 2017-08-17:
It may be worth maintaining a separate wiki page with information on camera models and their URLs.

lucaszanella, 2021-11-09:
Anyone knows how to access this camera via HTTP? I put the IP but I get

Not Found

The requested URL requested was not found on this server.

which indicates that the camera is responding. Here's the nmap scan:

nmap 192.168.1.22 -p 0-65535
Starting Nmap 7.80 ( https://nmap.org ) at 2021-11-08 22:33 -03
Nmap scan report for 192.168.1.22
Host is up (0.039s latency).
Not shown: 65532 closed ports
PORT      STATE SERVICE
80/tcp    open  http
554/tcp   open  rtsp
8888/tcp  open  sun-answerbook
10000/tcp open  snet-sensor-mgmt

I tried 8888 and 100000 but the camera does not respond in them

starbasessd, 2021-11-09:
Which camera? Brand and model?

starbasessd, 2021-11-09:
Most cameras URLs can be found at the iSpy database:
https://www.ispyconnect.com/sources.aspx

lucaszanella, 2021-11-09:
Anram AR-W620. This one: https://www.aliexpress.com/item/1005001586117737.html?spm=a2g0s.9042311.0.0.32944c4deStl1p

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

lucaszanella, 2021-11-09:
I tried all the http ones but none of them worked. Some of the snapshot.cgi or snapshot.jpg returned a malform image which firefox can't open. But what I want is simple HTTP access. I saw a guy on youtube acessing it just by the IP, nothing more. And the port 80 is live an responds

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

starbasessd, 2021-11-09:
It is not listed (W620) at all on the iSpy database. If you contact Anran or the person who had the setup in the channel, they would probably be best. The only info I can find doesn't have anything but their apps. The data may be encrypted, or otherwise proprietary. Many cameras are like that, even if they say h.264 compatible.

starbasessd, 2022-01-19:
No response in 60 days, closing.

closed 2022-01-19 by starbasessd