motioneye-project/motioneyeos #997: How many USB WEBCAM could be support per RASPBERRY PI 3?

Lakerway, 2017-06-14:
I insert a local Pi camera and USB webcam works great, Thanks !

there are four USB ports on raspberry Pi 3,  it would be too heavy for raspberry pi 3 if plugging in 4 USB webcam and one local camera? any tried it?

TheAmazingJarvis, 2017-06-20:
I have run three USB cameras along with the Raspcam camera total of four (I ran out of USB cameras). I think one issue MIGHT be current draw from the USB ports. You MIGHT have to add max_usb_current=1 to /boot/config.txt

Lakerway, 2017-06-20:
Thanks a lot, very appreciate for your time and contribution!

cuteboi, 2017-07-04:
if power is your only real concern, add a powered USB hub, as suggested by RaspberryPi Foundation.

kan-uddf, 2017-07-04:
What resolutions and frame rates were you getting with those usb cameras?

with one i seem to have massive frame dropping and miss things in the
recordings with only one usb camera

TheAmazingJarvis, 2017-07-06:
> What resolutions and frame rates were you getting with those usb cameras?

To be honest I'm not sure. I was using motioneye to keep track of an elderly family pet.  So probably pretty low.  Sorry.

jasaw, 2017-08-17:
Closing issue due to inactivity.

closed 2017-08-17 by jasaw