motioneye-project/motioneyeos #1004: Error capturing first image - USB SDD

neonnexus, 2017-06-21:
Hi,

So I have been successful in setting everything up on my Pi3 B apart from saving the captured recordings to my external USB SSD. I have one USB camera connected to my Pi as well as the USB SSD. The SD card I am using is only 2GB which was all I needed for the motionpie image knowing that I would be using the USB SSD for saving the capture movie files. The motioneye gui shows the device in the storage section and I have selected it and left the default path to save the files as the root / folder on the drive. I did think that I might need to format the drive (which I have not yet done) as it is currently NTFS and 43% full. If not having formatted the drive is not the issue, then I am not sure what I am doing wrong?

Movies is set to ON, Motion detection is set to ON, working schedule is set to OFF
 
Any help would be greatly appreciated. I cannot attach the logs until I get home tonight so any initial thoughts would be appreciated in the first instance. Thanks again.

ccrisan, 2017-06-22:
You shouldn't need to format that. As long as it's visible in the UI, the partition has been detected. Normally motionEyeOS will mount it for you and everything should work just fine. Just make sure it's really mounted (using `df` or `mount` commands). The motioneye.log file might also provide some helpful insights.

neonnexus, 2017-06-22:
Thanks, I found the issue, too much power being drawn from the Pi. When I connected all the devices through a powered USB hub, the issue went away. I had thought that because it was an SSD and not an HDD, there would be enough power but I guess the combination of the camera and the drive was too much for the standard Pi power output.

ccrisan, 2017-06-22:
Then we can close the issue.

closed 2017-06-22 by ccrisan