motioneye-project/motioneyeos #1010: Raspberry pi zero W error: Starting wpa_supplicant: failed (brcmfmac)

discarn8, 2017-06-27:
Gents:

I bought the zero w and downloaded the latest motioneyeos-raspberrypi-20170326.img.gz image.  I extracted it, wrote the image to an SD card via Win32DiskImager and then using Notepad++ added the wpa_supplicant.conf file to the root (using unix LF EOL).

My wpa_supplicant.conf looks like this:
---------------------------------------------------------
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
        ssid="myssid"
        psk="mypassword"
        key_mgmt=WPA-PSK
}

Boot the zero w from this sd card and i see:
"blah"
"blah"
"Starting wpa_supplicant: failed (brcmfmac)"
"blah"
"blah"
"Rebooting"

then the zero reboots.  

I tried copying the bootcode.bin/fixup.dat/start.elf from the raspbian jessie lite image over to the sd card and trying again - same results.

I keep reading about a firmware update for the zero but have yet to actually find the source for it - is there such an animal and is that what is holding me back?

mauricecyril, 2017-06-28:
I'm getting the same issue when trying to run motionpi image on a piZeroW

snoygues, 2017-06-28:
Exactly same issue for me.
I try several "solutions" found on the net, but none works.

TheRandy, 2017-06-29:
I have the same exact issue.  Same hardware, same "failed" message and then a boot cycle.  My wpa_supplicant looks like this:
 
-----------------------------------------
update_config=1
 ctrl_interface=/var/run/wpa_supplicant

 network={
     scan_ssid=1
     ssid="MyActualSSID(spelled correctly and case sensitive)"
     psk="MyActualPassword(correct and case sensitive)"
 }
--------------------------------------------

What am I missing here?

laguten, 2017-07-02:
I have this is problem too ((

jasaw, 2017-07-07:
This is my /boot/wpa_supplicant.conf file, and it works for me.

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=AU

network={
        ssid="SSID"
        psk="Password_Text"
}
```

ch1x0r, 2017-07-07:
Based on raspbian I presume... or ?

jasaw, 2017-07-07:
@ch1x0r I'm not sure what you meant by based on Raspbian, but that's the wpa_supplicant.conf file that I have on my Zero W running latest version motioneyeos.

ch1x0r, 2017-07-07:
MotioneyeOS - is it based on Rasbian?
How are you controlling the 0w? Male to female USB adapter or expansion
board? or ?
Are you trying to ssh in?

jasaw, 2017-07-07:
@ch1x0r motioneyeos is **not** based on Raspbian. It's built from buildroot.
I only have a USB power cable into my Zero W, and I can SSH in without any problem.

ch1x0r, 2017-07-07:
Nice!
Now, I'm inspired. I have a 0w. I'll try it.

Get back to you soon.

ch1x0r, 2017-07-07:
@jasaw I'm going to try bootcode.bin fixup.dat and start.elf to see if that works - this is my current path. LOL Have not forgotten you.

pat-cam, 2017-07-11:
hi all, when I followed the directions on the site linked, my pi zero w booted fine

[Manually setting up Pi WiFi using wpa_supplicant.conf](http://www.raspberrypi-spy.co.uk/2017/04/manually-setting-up-pi-wifi-using-wpa_supplicant-conf/)

in the tutorial, it's recommended using notepad++ to set the text file's EOL standard to Unix type - not sure if that's what's causing the failure for others...

jasaw, 2017-07-11:
@ccrisan Just an idea to avoid EoL problems. At boot time, check the line ending of /boot/wpa_supplicant.conf file. Automatically convert it to unix line ending if needed.

ccrisan, 2017-07-12:
Yeap, we could do that.

discarn8, 2017-07-17:
UPDATE:

Whether 'twas the popping sound made by myself or using a differenrt WiFi router or an update in the most current version - I am not sure - but, we finally have success with motioneyeos and the pi zero w.

Thank you for those that replied and assisted.  For those that may need this in the future, and are doing this on a windows machine - here are the steps I took to get mine working (nothing that hasn't been explained before).

Things you will need first:
- [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/files/latest/download)
- The [latest motioneyeos](https://github.com/ccrisan/motioneyeos/releases) image
- [Notepad++](https://notepad-plus-plus.org/download/v7.4.2.html)

1. Download the lates motioneyeos image
2. Extract the image to your hard drive (using Winzip, 7Zip, etc.)
3. Write the image to your uSD card using Win32DiskImager
4. DO NOT EJECT THE SD CARD yet.
5. Open Windows Explorer and browse to your drive containing the uSD card, You should see files such as:

    bootcode.bin
    loader.bin
    start.elf
    kernel.img
    cmdline.txt

6. Right-click in the right window pane and select "New Text Document"
7. Now, right click on this file and rename it wpa_supplicant.conf
8. Once again, right click on this file and select 'Edit with Notepad++'
9. Make sure you set the EOL conversion to UNIX LineFeeds (Edit-->EOL Conversion-->Unix LF)
10. Paste the following contents into your blank file (obviously changing the ssid and psk to match yours):

country=us
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
 scan_ssid=1
 ssid="MyWiFiSSID"
 psk="S3cr3tp@$$w0rc|"
}

11. Save the file
12. Eject your uSD card safely
13. Insert this uSD card into your Pi Zero W
14. (Optional) Attach your USB camera
15. Apply power to the Zero
16. If you have a monitor attached to the Zero, you should see the wireless network service starting, wpa_supplicant being read and 'brcmfmac Done'
17. Shortly, you should see your IP address being displayed along with your gateway and DNS servers
18.  Using your favorite browser, browse to the IP address that was displayed and log in to Zero using the default motioneyeos credentials of: admin / [blank password]
19. You can shut down the motioneye os remotely by select the advanced settings-->shutdown

If you do not have a display attached to your zero, you can use nmap to scan your home network for any new devices / ip's.  Look for one labeled "(Raspberry Pi Foundation)"

I hope this helps

[Some information borrowed from [http://www.raspberrypi-spy.co.uk/2017/04/manually-setting-up-pi-wifi-using-wpa_supplicant-conf/](http://www.raspberrypi-spy.co.uk/2017/04/manually-setting-up-pi-wifi-using-wpa_supplicant-conf/)]

closed 2017-07-17 by discarn8

silversword411, 2017-07-21:
Thanks for the writeup, transferred to Wiki :)

jasaw, 2017-07-21:
@silversword411 You might want to clarify that those steps you've added only apply for Windows?
As for Linux, simply mount and copy the wpa_supplicant.conf to /boot partition, then unmount.

ccrisan, 2017-08-06:
https://github.com/ccrisan/thingos/commit/8556a04e97fd164b30c7aa0acc77e5ca429251de should fix wpa_supplicant.conf files created in Windows.

tomeejr, 2018-03-24:
"Starting wpa_supplicant: failed (brcmfmac).  
Configuring wireless network: no device.   (Why is it looking for a wired device on a pi zero w ?)
I also have this exact problem on both pi zero w and pi3 mod 1.2.   Have read the helpful hints, have watched dozens of videos, have reformatted and tried hundreds of times, have used notepad++ as well as notepad to add the wpa_supplicant.con file.  I used the same board, cable, sd card, router to load raspian and it does connect wirelessly there, just not motioneyes.  The pi3 will run motioneye on ethernet,  but not wireless.  The pi zero w just reboots itself over and over.  Any ideas ?  Thanks.

ccrisan, 2018-03-24:
@tomeejr 

1. The message says `Configuring wireless network: no device`. It's saying that the wireless device cannot be found (i.e. `wlan0` is not present, for some reasons).
2. The `no device` message also indicates that it didn't even reached the part where it would read your `wpa_supplicant.conf` file, so don't bother with that one.
3. It's looking for a wired device as a fallback in case the wireless connection fails (some users actually connect an external USB ethernet adapter to their Zeros)
4. What OS version do you run? What exact image file did you download for your Zero and Pi3 boards? If you can provide more details and logs, maybe we could help you.

tomeejr, 2018-03-24:
motioneyesos-raspberry pi 20180224, using etcher, notepad++, windows 7home premium 64bit.  They made the name NOOB because that is me.  Can I transfer a card already booted from a pi3 to a pizw, and vice versa ?

richardgetz, 2018-04-04:
I’m having the same issue using the same version on my pi zero w except I’m using a Mac to create my wpa_supplicany. I am using the “motioneyeos-raspberrypi-20180224.img.gz” image. Wireless device says no link and wired says no device as I don’t have an Ethernet adapter. Any suggestions or something you’d like me to try?

brianfmorris, 2018-04-12:
I'm using Mac as well Pi Zero W, image “motioneyeos-raspberrypi-20180224.img.gz” and getting same error as noted - cycling reboots following "Starting wpa_supplicant: failed (brcmfmac)"

I have had no problems with a Pi B+ and Pi 3 B+ (but I have used their ethernet ports)

What ip address can you wirelessly SSH into?

richardgetz, 2018-04-12:
I was using an ethernet adapter and got it to work but I also solved my wireless issue. Your wpa_supplicant.conf file should read as below if you are using WPA2.

` 
country=us
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
network={
    ssid="SSIDNAME"
    psk="PASSWORD"
    proto=RSN
    key_mgmt=WPA-PSK
    pairwise=CCMP
    auth_alg=OPEN
}
`

richardgetz, 2018-04-12:
I was using an ethernet adapter and got it to work but I also solved my wireless issue. Your wpa_supplicant.conf file should read as below if you are using WPA2.

country=us
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev  

network={
    ssid="NETWORKSSID"
    psk="PASSWORD"
    proto=RSN
    key_mgmt=WPA-PSK
    pairwise=CCMP
    auth_alg=OPEN
}

[wpa_supplicant.conf.txt](https://github.com/ccrisan/motioneyeos/files/1904431/wpa_supplicant.conf.txt)

brianfmorris, 2018-04-12:
I greatly appreciate the input, but I was not successful with that version.  I did complete overwrite and reimaged SD

```
country=us
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev 

network={
	ssid="network"
	psk="password"
	proto=RSN
	key_mgmt=WPA-PSK
	pairwise=CCMP
	auth_alg=OPEN
}
```

richardgetz, 2018-04-12:
Just to make sure. You are running a WPA2 network with AES correct?

brianfmorris, 2018-04-12:
WPA2 yes.  Setting is WPA2/WPS Mixed personal (personal versus enterprise, but I can't find distinction between them).   Linksys EA6500 router

[image]

richardgetz, 2018-04-12:
Re-image then try this and see what it says

country=us
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev

network={
ssid="NETWORKSSID"
psk="PASSWORD"
proto=RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
}

richardgetz, 2018-04-12:
Actually you may want to try this. I've never tried a mixed network so I'm trying to figure out what would tell the pi zero the correct setup.

country=us
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev

network={
ssid="NETWORKSSID"
psk="PASSWORD"
proto=RSN
key_mgmt=WPA-PSK
pairwise=CCMP
group=TKIP
}

brianfmorris, 2018-04-12:
failed and relatively quickly

richardgetz, 2018-04-12:
Both failed relatively quickly?

richardgetz, 2018-04-12:
Did they both still give the same error?

brianfmorris, 2018-04-12:
yes, still the "failed (brcmfmac)"

richardgetz, 2018-04-12:
Can you confirm what motioneye os you are using?

brianfmorris, 2018-04-12:
motioneyeos-raspberrypi-20180224.img

richardgetz, 2018-04-12:
If this doesn't work I'm stumped. I'm pretty sure the mixed network is throwing it off. Some other notes: make sure 2.4ghz is enabled, make sure you are re-imaging each time.

country=us
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev

network={
ssid="NETWORKSSID"
psk="PASSWORD"
proto=RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
}

brianfmorris, 2018-04-12:
I've been erasing with SD Card Formatter and re-imaging with Etcher each time.

will try those next -

brianfmorris, 2018-04-12:
Ah ha!
@richardgetz totally pointed me at the answer.

Yes my 2.4ghz was enabled, as is my 5ghz... what I completely failed to appreciate is that the Pi Zero W is not 802.11ac compatible (i.e. not 5ghz compatible).  In all my attempts, I kept pointing the Pi to connect to the 5G SSID, not my 2.4ghz SSID.

Thank you all!

jasaw, 2018-04-13:
Anyone who's having wifi connection issues might want read the [wireless troubleshooting guide](https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems) first. It may save you a lot of time and frustration. The lack of 5GHz wifi support on Pi Zero W is covered in the wiki too.

HOD42, 2018-07-05:
I had a similar problem. The solution appears to have been modifying my initial wpa_supplicatnt.conf to include DIR and GROUP as follows:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
Also the addition of:
country=GB

N59175, 2020-01-12:
I've done all this and tried another router, still no dice.  Using motioneyeos 20190911.
raspberry pi zero w works perfect on Raspian, I just want to use Motioneyeos.  I cannot get this thing to connect to a router.  crazy

N59175, 2020-01-12:
It connects if I leave the security off

N59175, 2020-01-26:
I got it finally!!  I connect to a RPI3 via wired network, configure it to use wireless, test it then retrieve the wpa_supplicant file it created from the sd card, copy that file to the RPI zero w and it works!!  no idea why.  The formatting must be different.  regards

DrCR77, 2020-06-27:
I have already 3 Pi working (2 pi zero W) with motioneyeos and would like to add a forth.
Strangely I ran into this exact same problem:
I tried: 
2 different pi zero (W 1.3 and WH)
2 different SD cards
3 different chargers
image 20190911 and 20200606
tried copying the wpa_supplicant.conf manually (via mac) or using writeimage + network settings on linux.
also checked followed the trouble shooting page
Nothing has worked so far.

The wierd thing is that it worked for a few days on the exact same zero WH before until it stopped. Tried to freshly set it up in the same way failed. Then tried another brand new pi zero W and all sorts of variations of images, SD cards etc.. 
Even changed the wpa_supplicant details several times including EOL. 
nothing brings me beyond the point: 

starting wpa_supplicant.conf = no device
==> bootloop 

I can connect via HDMI. So I tried to disable wifi in config.txt to perhaps set it up manually after the first boot but then the sequence breaks on the first boot at network settings 
Is there any help on this?

starbasessd, 2020-06-27:
Can you run a straight Raspberry PI OS on it and have the WiFi work? I had
one where the wifi chip failed, and presents as yours does.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DrCR77, 2020-06-27:
Thanks for the response! It seems just to be a hardware issue. The wifi was corrupted on the Zero WH and it turned out that the Zero W was actually just a Zero without any "W"...

starbasessd, 2020-06-27:
It happens.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?