motioneye-project/motioneyeos #1011: Add an Arm/Disarm feature

JFCCoding12, 2017-06-27:
Add a feature so the user can arm the camera and allow notifications and movies to be enable. And have a button so the user can disarm it. So the user is not always recording or getting notifications when they are home.

jasaw, 2017-07-18:
You can arm/disarm it manually by logging in to motioneye web UI, and enabling/disabling the motion detection.

Would be better if the arm/disarm is automatic. If a phone is connected to the wifi network, disarm it. If none of the specified phones are connected, arm it.
Use case is, if any of the family members are at home (at least one phone connected to wifi network), disarm it. If no one is at home, no phone is connected to the wifi network, arm it.

I imagine this would be the most common use case for most people.

jasaw, 2017-07-31:
There has been some work done by other users that you may find useful. Refer to ccrisan/motioneye#203.