motioneye-project/motioneyeos #1012: Pi Zero WiFi not working after first reboot

stevexz, 2017-06-29:
Hi,

using the standard procedure wpa_supplicant.conf to add WiFi for the Pi Zero.
Works as intended at first boot and everything seems ok.

But when I from the web admin interface makes some changes like change of time zone and then reeboot I get Starting wpa_supplicant: failed

Tried with both stable and latest.

ccrisan, 2017-06-29:
How does your `wpa_supplicant.conf` look like before (i.e. the one you supply) and how does it look like after the changes (`/data/etc/wpa_supplicant.conf`)?.

stevexz, 2017-06-29:
Before:
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
     scan_ssid=1
     ssid="bahnhof2_4Ghz-206052"
     psk="U9Ran7KSWsUdr"
}

After change(before reboot, with nano):
update_config=1^M
ctrl_interface=/var/run/wpa_supplicant^M
^M
network={^M
     scan_ssid=1^M
    ssid="***********"
    psk="**********"
}

ccrisan, 2017-06-29:
This doesn't make too much sense. The `^M` is something that windows machines would add. However they shouldn't affect the file. Besides that, the file looks identically to me.

stevexz, 2017-06-29:
The situation with the reboot is the same as if you at first boot did forget the wap_supplicant and add it before next boot.

flatlining, 2017-07-12:
Same problem here, I notice that the */etc/wpa_supplicant.conf* file on the partition created on the remaining space (not boot nor root) **key_mgmt=NONE** is added to the network and if I deleted the zero w connects again.

But any change done (e.g.: timezone or password) were not saved, and if I try again it happens again (**key_mgmt=NONE** added to the file).

Hope I helped somehow.

jasaw, 2017-07-14:
Interestingly, I'm seeing this issue happening on my 2nd router. It keeps adding **key_mgmt=NONE** just like @flatlining is seeing. Works fine on my 1st router.

jasaw, 2017-07-14:
Very interesting... I found a workaround to prevent it from adding the key_mgmt=NONE by adding another "network" block to confuse it. Make sure your actual network block is the last one in the file.

Example:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=AU

network={
        ssid="Dummy SSID"
        psk="Random Text"
}
network={
        ssid="Actual SSID"
        psk="Actual Password_Text"
}
```

Start with a fresh image, copy this new wpa_supplicant.conf to your boot partition and let it boot up. See if this works for you guys.

ccrisan, 2017-07-14:
Guys, can I have a wpa_supplicant.conf file *before* it gets the `key_mgmt=None` line? Also I'd need to have some insights on the ssid and psk. Normally, the `key_mgmt=None` is added when no psk is supplied (assuming an open network).

jasaw, 2017-07-14:
Actual file:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=AU

network={
        ssid="Dummy Wifi"
        psk="Dummy Password"
}

network={
        ssid="DarkNet"
        psk=3d04392b31763f7c7212f158067b1eaea1706ccef18ce6c96eb392b1e8eb3241
}
```

ccrisan, 2017-07-14:
@jasaw Is this the file that is modified with `key_mgmt=None` by motionEye? Or is this the one that works?

jasaw, 2017-07-14:
That's the file that works. Just remove the "Dummy Wifi" network block and it stops working.

ccrisan, 2017-08-05:
https://github.com/ccrisan/motioneye/commit/e895ff48ee50ff4c4e79803999b222855809c6c6 fixes this by adding support for hex-encoded PSKs.

closed 2017-08-05 by ccrisan