motioneye-project/motioneyeos #1013: Mech. IR-CUT Filter uses Camera Led I/O to switch

bigplums, 2017-06-29:
Hi,
I have [this camera](https://www.modmypi.com/raspberry-pi/camera/camera-boards/raspberry-pi-night-vision-camera-ir-cut/?limit=50)  which uses the camera on led I/O to actuate the mechanical IR-CUT filter.
I wanted to know if there was anyway to be able to have this automatically switch over to 'Night mode' (led I/O = LOW) without any human intervention when the camera is in low light?
Many thanks in advance
Ross

jasaw, 2017-07-18:
@bigplums I'm interested in trying out the same camera that you have. Optically detect light level and controlling the IR-cut filter via camera red LED would be an elegant solution if achievable.

First, we need to workout, on the Pi Zero W, which GPIO controls the camera red LED. There's some information about controlling the camera LED [here](http://www.raspberrypi-spy.co.uk/2013/05/how-to-disable-the-red-led-on-the-pi-camera-module/), but no info about Pi Zero W. Controlling the LED via GPIO is simple.

Second, we need to modify motion software to optically detect light level. This is the difficult part. How do you optically detect "day light" reliably? The problem is, we need to detect "day light" by filtering out infra-red because we have IR LED light for "night vision". We don't want to false detect this IR LED light as "day light" and activate the IR-cut filter at night. Is comparing the red channel in the image against blue and green going to be reliable enough? (red sensors in camera are most sensitive to IR) Maybe a much easier solution is to add an LDR like this:
[image]
Notice the LDR has an IR-cut filter?

bigplums, 2017-07-18:
In the end I ended up keeping it simple and just used a CdS cell, transistor and resistor to trigger the IR-CUT Filters I/O which is shown [here](http://www.waveshare.com/wiki/File:Rpi-ir-cut-camera-faq-4.jpg)

It would be nice if there was some built in way of triggering the Pi I/O on an event in MotionEyesOS I would think it could have quite a few handy uses - maybe thats something for the Devs to mull over one day if they get time from fixing the more important bits first!

Ross

charemer, 2017-08-04:
Landed here looking for a solution to the same problem - Being able to use camera action buttons to manually switch to/from IR mode would be a start.  It's all dependant on being able to map a GPIO on the Pi Zero to the camera LED.  Has anyone else tried this on a Pi2 or Pi3 while running Motioneyeos?  Failing that I was looking at the IR led board circuit and was wondering if there was suitabl point on that board to take a logic signal from to patch over to the camera I/O pad.  @bigplums  could you post a link to a schematic for the circuit you used as a workaround please?

jasaw, 2017-08-04:
Camera LED can be controlled from software. This may be too technical for some, so a pull-request would be nice. :-)

**Pi-B**
```
echo 5 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio5/direction
echo 1 > /sys/class/gpio/gpio5/value
echo 0 > /sys/class/gpio/gpio5/value
```

**Pi-B+, Pi2**
```
echo 32 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio32/direction
echo 1 > /sys/class/gpio/gpio32/value
echo 0 > /sys/class/gpio/gpio32/value
```

**Pi3**
Control via the Mailbox interface (/dev/vcio)
This [C program](https://github.com/6by9/rpi3-gpiovirtbuf) can be used to control the camera LED.

**Pi Zero W**
I haven't found any information, but from the schematic, it looks very similar to Pi3, so I would try the Pi3 method first.

charemer, 2017-08-07:
I've found the answer the Pi Zero W camera led can be controlled from gpio40.

    echo 40 > /sys/class/gpio/export
    echo out > /sys/class/gpio/gpio40/direction
    echo 1 > /sys/class/gpio/gpio40/value
    echo 0 > /cys/class/gpio/gpio40/value

jasaw, 2017-08-08:
@charemer Thank you for sharing the information.

SuneBielefeldt, 2017-09-09:
It is now possible to buy the Raspberry Pi Camera with IR-CUT filter that automatically switches to night mode when it gets dark.

It eliminates the need to manually control the IR-CUT filter.

See the inserted link.

[https://www.aliexpress.com/item/Latest-Raspberry-Pi-3-Model-B-IR-Cut-Camera-5MP-Night-Vision-Automatic-Day-and-Night/32831967873.html](url)

[image]

SuneBielefeldt, 2018-01-22:
Hey Davimas
No, I do not know other MotionEye forums than this, but it works well too. 

Have now used the ir cut camera for some months and am fully satisfied with it.

davimas, 2018-01-23:
Hey, Thanks.
I bought Yesterday in AliExpress this IR CUT RaspiCam. 
When you use it in Motioneye with MDetection, you became a Alarmmessage when the Cam Switch the Filter? I think yes, every picture change send me a Message.......

jasaw, 2018-04-27:
@EpicLPer Using the camera image to determine when to enable/disable the IR-cut filter would be very tricky (assuming you have IR lights) because the IR light will interfere with day-light detection. Read my earlier comments for more details.

I'm also using this camera with manual IR-cut filter, and I've built a simple light-dependent resistor (LDR or CdS) with [small IR-cut filter](https://www.ebay.com.au/itm/10pcs-Optical-UV-IR-Cut-Light-Filter-Blocking-For-Camera-Astronomy-Lens-8mm-Slim/282653207704) to detect light, and some IR-LEDs for illumination. I've also written a simple [LDR reader](https://github.com/jasaw/ldr-reader) program that reads the LDR and sets GPIOs to high/low and runs commands. It has been working well for me.

charemer, 2018-04-27:
> Hi, I want to bump this a little. The solution from @charemer [1] worked perfectly so far but has to be done manually, obviously. An Expert option where you can enable when the picture gets too dark to run a command and vice versa when it gets bright again would be a real awesome thing for this! :)

I used incron and the program remind to automate setting dusk and dawn
times each day at which to switch the IR filter. 

One job runs just after noon each day to set the following dusk dawn
cycle. by creating new at tasks. 

  

Links:
------
[1] https://github.com/charemer
[2]
https://github.com/ccrisan/motioneyeos/issues/1013#issuecomment-384594160
[3]
https://github.com/

ctschach, 2019-02-10:
Okay, I would like to step in here again. I also would love to see native night vision support. 

So I compiled "rpi3-gpiovirtbuf" on a different Raspberry and copied it over to the MotionEyeOS machine to "/data/bin". When I run:

```
[root@meye-5cbab7a1 bin]# ./rpi3-gpiovirtbuf 134 0
Get state of 0 as 1
get_config dir input, active high, pulled high
```

It looks as expected. However, nothing happens to the camera LED which triggers the IR cut filter. But if I change the LED status inside of the MotionEyeOS Backend, the switch work.

So what is the trick to get this running?

fschlaef, 2023-12-28:
> I've found the answer the Pi Zero W camera led can be controlled from gpio40.
> 
> ```
> echo 40 > /sys/class/gpio/export
> echo out > /sys/class/gpio/gpio40/direction
> echo 1 > /sys/class/gpio/gpio40/value
> echo 0 > /cys/class/gpio/gpio40/value
> ```
> 
> I should add a thankyou to @jasaw for posting the Pi3 answer which led me to the dt-blob.dts file.

Unfortunately, for the Pi Zero 2 W, Pin 40 is now "Camera shutdown" .
https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts has the following in the "pins_pi02w " section

`            pin_define@CAMERA_0_LED {
               type = "absent";
            };
`

Unfortunately, this is no longer an option :(