motioneye-project/motioneyeos #1014: Android app unavaliable in Brazil

hpicelli, 2017-06-29:
Hi there.
I can't install the android app, in my country it is unavaliable.
But, i can install directly the .APK.
What version is the latest?
Best regards,
Henrique
[image]

ccrisan, 2017-06-29:
We don't have an android APK.

closed 2017-06-29 by ccrisan

hpicelli, 2017-06-29:
Thanx. I guessed tha have a client.
[image]

Garskecom, 2017-07-14:
Hi! We've made a completely new version - Please search for MotionEyeOS on the App Store again to find it.