motioneye-project/motioneyeos #1016: Problems booting on R-Pi3

JohnMAndre, 2017-06-30:
I'm quite new to using a Raspberry Pi but I have successfully installed and used two different OS's (Raspbian and OSMC). 

I downloaded the latest stable build of motioneye (motioneyeos-raspberrypi3-20170329.img.gz), wrote the image to a 32GB new SD card, and tried to boot it up. It powers up, there is a long list of things it is doing (I guess the initiating sequence). Then it hangs for about 2-3 seconds. Then it reboots. It will continue this cycle forever, every time rebooting in the middle of the boot sequence.

I have also tried with motioneyeos-raspberrypi3-20170610.img.gz and that was no better.

I do not see others having any problems so could I be doing something silly that everyone else knows to do? I am using a R-Pi3 and everything else (including the SD card) is the same as what works for OSMC.

ccrisan, 2017-06-30:
Without telling us what the boot sequence complains about, we can't help you.

cityguy0002, 2017-07-02:
Sounds familiar to me. This happened on my installation while booting without an active network-connection (wired or wireless). Try to boot wired first.

jasaw, 2017-07-24:
@JohnMAndre Please connect a monitor to your Pi, watch the boot sequence text, and tell us where it got stuck.

Might be a lack of network connectivity issue, like what @cityguy0002 mentioned.

JohnMAndre, 2017-08-15:
I finally found some time to dig into this issue. Here's what I discovered:

If the Pi is not connected to ethernet, MotionEye will reboot without end. I could not see the text before because of the small font (was connected to a TV). After getting some extra hardware, I could see the text which, combined with @cityguy0002 's helpful comment got the fix.

It would be great if it would stop and instead of just rebooting forever if it would say **"An active network connection is required to boot MotionEye"** or similar.

It recognized a webcam rightaway - great!

Then, I entered the wifi information, since Ethernet is not great for me, and it rebooted...and rebooted...again and again, without end. 

I figured I must have mis-entered the wifi key but I could not see how to re-enter it. Even plugging back in the Ethernet, it would reboot every time it would see the problem with wifi.

It would be great **if wifi fails, it could fall back to ethernet** so someone can still boot and fix the problem. I had to re-image my card to fix this issue.

It would also be great **if it could prompt for user-input to set the wifi** (so no ethernet is needed even on the initial boot).

ccrisan, 2017-08-15:
@JohnMAndre, a camera firmware is not supposed to prompt the user for anything. Printing messages in log files or on the monitor is useless for regular users. When things go wrong (a temporary connection issue), you expect it to reconnect instead of simply printing an error.

Reconnection is more efficiently done via a reboot. In case there's a hardware/driver issue that caused the problem, it's more likely for a reboot to fix it.

Cameras very rarely have two connections; implementing a fallback for when one connection fails has very little impact.

This being said, once you have the camera in place and everything works as expected, in my opinion, the best "unattended" recovery is the way implemented by motionEyeOS these days.

closed 2017-08-15 by ccrisan

JohnMAndre, 2017-08-16:
@ccrisan I see that a lot of work and thought has gone into the creation of motionEye and I'm thankful for it. I do think setup issues are different from operating issues. It seems like a lot of attention has gone into operating issues but more could be done to make initial setup easier.