motioneye-project/motioneyeos #1017: :: Unable to upgrade - still ::

rafi1975, 2017-06-30:
HI,

Still having issues with upgrades:
the same error - INFO: updating to version 20170611...
2017-07-01 00:42:53: [motioneye]     INFO: killing motioneye init script...
2017-07-01 00:42:53: [motioneye]     INFO: stopping netwatch init script...
2017-07-01 00:42:53: [motioneye]     INFO: downloading https://github.com/ccrisan/motioneyeos/releases/download/20170611/motioneyeos-raspberrypi3-20170611.img.gz...
2017-07-01 00:42:54: [motioneye]    ERROR: could not download update: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motioneyeos/releases/download/20170611/motioneyeos-raspberrypi3-20170611.img.gz', '--no-check-certificate', '-O', '/data/.firmware_update/firmware.gz']' returned non-zero exit status 8
2017-07-01 00:42:54: [motioneye]    ERROR: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motioneyeos/releases/download/20170611/motioneyeos-raspberrypi3-20170611.img.gz', '--no-check-certificate', '-O', '/data/.firmware_update/firmware.gz']' returned non-zero exit status 8
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi3/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1443, in _execute
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 175, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 1787, in post
  File "/usr/lib/python2.7/site-packages/motioneye/update.py", line 199, in perform_update
  File "/usr/lib/python2.7/site-packages/motioneye/update.py", line 132, in download
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call
CalledProcessError: Command '['/usr/bin/wget', u'https://github.com/ccrisan/motioneyeos/releases/download/20170611/motioneyeos-raspberrypi3-20170611.img.gz', '--no-check-certificate', '-O', '/data/.firmware_update/firmware.gz']' returned non-zero exit status 8
2017-07-01 00:42:54: [motioneye]    ERROR: 500 POST /update/?version=20170611&_=1498866170848&_username=admin&_signature=a4430cb87717292cb5f824a36acae29d36f160de (192.168.1.167) 678.49ms

All cameras have Internet access - defo. They can send email notifications ok.

Thanks,

Rafe

jasaw, 2017-11-14:
Which version of motionEyeOS are you running? What hardware configuration?

jasaw, 2019-07-12:
Closing issue due to lack of response from issue author.

closed 2019-07-12 by jasaw