motioneye-project/motioneyeos #1019: No preview after login into admin panel

k0b1, 2017-07-05:
Hi, 
I would like to say Thank you for superb job! 

Since last week I'm runing my first cctv camera based on your motioneyeos and I'm very happy from results but I have one problem.
I'm trying to setup motion detection, for that I need image from my camera, problem is that: once I will log in to admin preview is gone. If I would't login to admin panel all is fine, can see preview window and basic preferences on left + two top icons. 
I have try firefox, chrome, ie  and no positve resutls.
My camera stream to web :8081 and this stream goes on DS synology station without any problems also if I will go stright to ip:8081 preview of stream is ok.

Please advice.
Thank you

ccrisan, 2017-07-05:
Please post a screenshot of the problem. Normally the port 8081 isn't involved directly anywhere in this scenario. You should only use the web frontend which runs on port 80.

k0b1, 2017-07-05:
thank you for your quick respond! 
[image]
[image]

k0b1, 2017-07-05:
one more thing, because my sd card starts to be filled up with some video files ( I'm using synology to keep all my files..) I have changed inside file storage - to custom patch /dev/null ( yesterday this cause problem on 8081 streaming website - after 1-2 sec picture got frozen. Today all is fine :) somehow :)

ccrisan, 2017-07-05:
Set the framerate dimmer and resolution dimmer prefs to something greater than 0.

closed 2017-07-05 by ccrisan