motioneye-project/motioneyeos #1021: Low Frame rate on Pi Zero W

adrian0, 2017-07-06:
Hi,
Even when configured for 30fps (Movies and streaming), I am unable to get better than about 0.2fps
This is running a resolution of 1280x1024 and overclocking to 1000Mhz.

Anyone know what I can do to get a more reasonable frame rate?

Will a full blown Pi3 provide better frame rate than a Pi Zero W?

Thanks,
Rock hard, Rock often.
-Adrian

jasaw, 2017-07-06:
@adrian0 The reason you're getting low frame rate is because motion software by default does **not** use hardware acceleration when encoding movies.
Work has been done to enable hardware acceleration but not yet merged into master. Refer to issue #365 and pull request 998 for more information.
I'm running Pi Zero W with hardware acceleration, gettting 25fps, 3.5Mbps at 800 x 600 resolution. If you want to try out the binary that I've built, go to my [fork](https://github.com/jasaw/motioneyeos/releases), but bewarned that it's highly experimental, that's why it's not merged into master yet. Use at your own risk.

adrian0, 2017-07-07:
Thanks, jasaw,
Awesome information and help.
I will test out your fork over the weekend.

Ball park (I know how these things are), at what point do you feel this will be merged into the main branch?

Thanks,
Rock hard, Rock often.
-Adrian

jasaw, 2017-07-07:
@adrian0 Sorry, that question is for @ccrisan to answer. He's the project owner. I'm just another motioneyeos user. :-)

adrian0, 2017-07-07:
"..just another motioneyeos user" :-)
Understatement!

Thanks for making the motioneyos world a better place for us less technical people.
(And thanks to @ccrisan for the project)

ccrisan, 2017-07-07:
@adrian0 we're actually waiting for the next motion release, which is already being discussed and which comes with many ffmpeg-related fixes. I believe this makes sense since @jasaw's work is related to ffmpeg.

dchang0, 2017-07-08:
I notice that my Raspberry Pi Zero W in Fast Network Camera mode also produces very low framerates. Is it true that streamEye in Fast Network Camera mode also uses ffmpeg, or could this be some other problem?

ccrisan, 2017-07-08:
@dchang0 , Fast Network Camera does **not** use ffmpeg. It should stream as high as possible, given the streaming format (MJPEG) and the network bandwidth. What are your settings (resolution, framerate and quality) and what is the approximate actual framerate that you get?

dchang0, 2017-07-08:
Thanks for the quick reply. I won't dirty up/split this thread with my issue. Now that I know it's not ffmpeg, I'll look for other causes and start a separate thread once I've gathered enough diagnostic info.

dchang0, 2017-07-09:
I'll ask this question here in this thread because it is pertinent to both my issue and this one: is there a utility that can measure actual framerate of a received IP camera stream and optionally graph it over time?
The reason I ask is that my current surveillance software says my actual framerate is a lowly 1fps, and I want to double-check this value (it certainly seems to be correct from viewing the video).

Reducing the resolution does visibly improve framerate, so I guess the goal is to measure the maximum that the RPi Zero W can currently do and then work from there to get it on par with the other RPi models that are supposed to have similar hardware capabilities.

jasaw, 2017-10-12:
My hardware accelerated video encoding work is in pre-release 20171008.
It should address the low framerate issue for those recording video locally.
If you're getting low framerate with fast network camera mode, then it's most likely wifi signal issue.

closed 2017-10-12 by jasaw

jeffehobbs, 2019-01-19:
Checking into this thread in 2019, I've installed motionEyeOS on a Raspberry Pi Zero W, followed [this](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) (excellent!) walkthrough for setup, but I too am getting low 1-2fps frame rates from captured video. Is there some setting or config pref that needs to be set OOTB to get better performance from the Pi Zero?

jasaw, 2019-01-21:
@jeffehobbs Did you follow the motionEye on Raspbian installation wiki or did you install motionEyeOS? If you are running motionEyeOS, you have to select the OMX encoder under Movies section to get higher frame rate.
If you are running Raspbian, you'll need to do more things to enable OMX. See https://github.com/ccrisan/motioneye/issues/930

jeffehobbs, 2019-01-21:
Thanks, @jasaw. I've installed motionEyeOS, most of the settings are OOTB defaults, and I've got movie encoding set to "H.264/OMX" in that dropdown, but I am still seeing very rough frame rates. Screenshot below:

[image]

To be clear, I need motion detection in my project so "[Fast Network Camera](https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera)" mode is not an option for me. But I am interested in getting higher frame rates from the the Pi Zero W.

jasaw, 2019-01-21:
@jeffehobbs What's your resolution and frame rate? Pi Zero W hardware is pretty weak compared to the big brother RPi 3B+.
From memory, I think Pi Zero W is only capable of doing something like this:
- 800x600 @ 25 fps.
- 1024x768 @ 20 fps.
- 1280x720 @ 15 fps.

You'll also need to close your web browser to stop the live video stream. Live video stream puts heavy load on CPU.

Snipercaine, 2019-02-17:
I'm also getting ~2 fps, on my PiZeroW with MotionEyeOs installed.  I'm using H.264/OMX (.mp4), and 1280x720 15fps.  But the stream is showing 2 fps.  
Do we have a clear solution for this?

jeffehobbs, 2019-02-17:
I found that changing rotation and _actually rotating the physical camera_ helped a bit, but I echo @Snipercaine's comments above -- frame rates on the Pi Zero W are still very low.

jasaw, 2019-02-17:
@Snipercaine Live streaming will give you very low frame rate because it uses the CPU to encode the mjpeg stream. The 15 fps can only be achieved when recording video using h264/omx.