motioneye-project/motioneyeos #1026: Busybox timeout

Chiny91, 2017-07-09:
It would be useful to have Busybox include timeout, which is currently `# CONFIG_TIMEOUT is not set`.

The standard rsync (so useful for backups, storing pictures elsewhere) can be called, usually with cron, but without fancy script work it is uncertain whether rsync is already running.   rsync can be compiled with an option for `--time-limit` or `--stop-at` but that is more complex albeit useful.     Using cron to call rsync with a timeout (before the next cron) would do the job nicely.

ccrisan, 2017-07-09:
https://github.com/ccrisan/thingos/commit/9b2e285c38f741fe321b150fe23d36999676d920 adds support for the `timeout` command.

closed 2017-07-09 by ccrisan

Chiny91, 2017-07-14:
Excellent; and this update included within `motionEyeOS 20170710` has been used as my above post for 24 hours and is working well.

For the info of anyone thinking of using timeout, note this is the `BusyBox `version, not the `coreutils` version and has different syntax - from [the BusyBox words](https://busybox.net/downloads/BusyBox.txt):
```
timeout [-t SECS] [-s SIG] PROG [ARGS]

        Runs PROG. Sends SIG to it if it is not gone in SECS seconds.
        Defaults: SECS: 10, SIG: TERM.
```
In summary, rsync has been providing ultra-reliable backups from 2 `motionEyeOS` setups, for many months and the addition of `timeout` should avoid a potential problem.