motioneye-project/motioneyeos #1027: Saving files to Google Drive stopped working

onlize, 2017-07-10:
Everything worked for me for some time. Now, I found that movies are no longer saved on Google Drive. When I click on button to test service, the message tells me that the test was successful, but files are not saved.

What interesting is that on Google Drive new folder is created every day, but no files in the folder.

I tried to reboot it a couple of times, but no luck.

Can you help?
Thank you.

ccrisan, 2017-07-11:
Does your `/data/etc/uploadservices.json` file contain an entry for the `gdrive` service? **don't paste it here**, as it contains some sensitive data.

onlize, 2017-07-11:
Yes, it has the following:
{
    "1": {
        "gdrive": {
            "authorization_key": "####################"
            "credentials": {
                "access_token": "####################
                "refresh_token": "####################
            },
            "location": "####################"
        }
    }
}

I believe that it can connect to Google Drive because it creates folders, but not files.

Thank you.

ccrisan, 2017-07-11:
It looks fine. Do you get any specific error in your motioneye log file?

onlize, 2017-07-11:
Sorry, but where is the log file located?

Thank you.

Chiny91, 2017-07-11:
Log in (I suspect as admin is required), enable `Expert settings` if you have not already, scroll to near the bottom of expert settings and there you are... click on the log required.

onlize, 2017-07-12:
Found it. The log helped me. 

Thank you for your help.

closed 2017-07-12 by onlize