motioneye-project/motioneyeos #1029: Feature request: usage bar (CPU etc or effective aggregate)

Jodes81, 2017-07-11:
I am struggling to know where bottlenecks are as I try to get higher frame rates. I want a way of seeing quickly, while I change parameters, whether the machine running MotionEyeOS is in some way the limiting link in the chain.

For example, I'm using a Raspberry Pi 3 to record a stream from an IP camera. I'm getting really poor frame rates. I don't know if it's CPU, disk access or whatever. I woukd like a live graphical bar that shows instantaneous usage within the web interface. Either one bar for each stat, or one which shows the maximum of all combined to indicate that any one limit or another is being reached.

I am not good at linux and really appreciate the ease of use of the OS. A usage bar would be useful, so I won't have to SSH in, battle with htop commands etc, or generally get embedded brain ache.

Chiny91, 2017-07-11:
Sounds fun, although using such a tool inevitably stresses the R Pi 3 even more 😄    On my R Pi 3s, I found the cpu groaned first with 3 IP cams, although even with the res turned down, the ethernet could struggle.   I ended up with all 3 cams (per R Pi 3) set at 704*576 and 4 frames/second which has proved rock stable.   This would be disappointing to many, no doubt.

Jodes81, 2017-07-11:
I'm using it for security, and need it on its highest resolution (720p) which is barely enough as it is, and I'm only getting up to 10fps which is OK.  But I hoped to add a second camera at 1080p, 10fps or higher. From what you say, it definitely won't cope. So I'm considering buying the most powerful boards that MotionEyeOS supports instead, possibly with ethernet switches, or if necessary, one board per camera. But whatever happens, the goal is low cost - which is why I'm not just hooking up a fully fledged PC/server, as the energy bill is unaffordable! Plus it's hard to know what would have enough processing power anyway.

But back on topic, the resource meter could be activated or deactivated, so not always using resources? And/or it could just log usage for viewing separately? But I wouldn't have thought using ajax for updates would be heavy at all, on the RPi at least, just sending mere bytes at a time. Although I don't know how expensive it is to obtain real time usage data?

jasaw, 2017-08-17:
I've been monitoring CPU, GPU, and memory usage manually via ssh. I wrote scripts to poll these information, but find that it's a pain to keep watching the terminal.

I think it would be nice to be able to turn on "system health monitor" from the web UI, sacrifice a bit of CPU cycles and some memory, to get visibility into how the system is coping with given configuration (resolution, frame rate, etc). Graph the CPU load over the last x minutes, track the max GPU and CPU memory usage, and OOM events would be very helpful in finding an optimum configuration for a given set up. When the user is happy with the configuration, "system health monitor" can be turned off.

We should perhaps consider implementing this.

ccrisan, 2017-08-17:
We have the monitoring commands and we have the RPi system monitoring wrapper that displays stuff like sysload or cpu usage on top of the camera frame. Isn't that enough for our needs?

jasaw, 2017-08-17:
This is about storing and visualizing historical system stats like the max memory usage over the uptime of the device, GPU memory usage, CPU load of the last few hours. Temperature as well. If there's OOM events, we want to know about it too. 

Not sure how difficult to implement though. 

I guess this is a nice to have feature that may come in handy at initial install and test stage. Definitely not high priority.

ccrisan, 2017-08-17:
Ah right, I didn't read this through. Well in that case, I don't see this implemented anytime soon. It's a nice-to-have feature, but definitely not something as useful as performance when encoding and browsing media files :)