motioneye-project/motioneyeos #1032: Switch to lan instead of wifi

pauloreal1, 2017-07-12:
HI, my wifi card doesnt connect (turn on) in raspberry with motioneye os, it was working fine and it works in my pc. 
How do i change to lan if i cant access with the wifi? Is there any way to do it in config or something? 
Thank you

ccrisan, 2017-07-12:
Just plug in an ethernet cable.

closed 2017-07-12 by ccrisan

pauloreal1, 2017-07-12:
Ok, i was leaving home in a hurry so i didn't even connect to HDMI to see if it was some error on card maybe. If the system was booting.
I did connect the cable with no luck, so it must be a corrupt card or something. Anyway the wifi adaptor isn't turning on, so it could also be some kind of power issue. I just didn't had the time to check it properly so i asked you to dismiss any other issue before I get home later. Thank you.